360Controller/360Controller #602: Not working with high sierra

jldav, 2018-02-10:
Xbox 360 Controller won't connect to wireless adapter. Adapter shows up under systems in Mac, just updated to high sierra and the xbox360 controller just blinks and won't connect.

jldav, 2018-02-10:
Just tested this on my work laptop. Running Yosemite, same controller and same wireless adapter. It works in Yosemite. So this driver does not support high sierra, at least not for me.

StTheo, 2018-02-11:
I had issues after restarting until I checked "Security & Privacy" and allowed the "Drew Millis" app manually. Then the light turned on and the preference pane knew about it.

jldav, 2018-02-11:
Thanks for the tip, I allowed the Drew Millis app but it still won't connect :(. If anyone has had any luck with High Sierra let me know! Some of my Steam stuff won't play on Yosemite, but I can't get the controller to work with High Sierra still.

Cruplezone, 2018-02-11:
You really shouldn't be using the wireless adapter with this driver, it is known to reliably cause kernel panics on sleep.

FranticRain, 2018-04-02:
Wireless adapters not working in Sierra is a known issue caused by a rewrite of the macOS USB kernel introduced in 10.11. Discussion consolidated in #605

closed 2018-04-02 by FranticRain