360Controller/360Controller #603: Wired Xbox One controller not connecting

SupremeSalamander, 2018-02-14:
I dont know if this is a problem with the software, my computer, or my controller. I have plugged in my controller to my iMac, but the controller's light keeps flashing, signaling that it isn't connected. When I open 360Controller, the program doesn't recognize it either. Please give me some tips.

MaddTheSane, 2018-02-14:
Is the cable a charging cable? If so, you need a cable that can also transmit data.

SupremeSalamander, 2018-02-14:
It's not a charging cable. When i plug it into my xbox it works fine, but it doesn't work on my computer.

arthas994, 2018-02-21:
Check if under "Security & Privacy" in system preferences "Drew Milis" App is allowed.

nalancer08, 2018-03-26:
@arthas994 Thank you so much!!! This step its necessary in intilling guide.

lilphil123, 2018-03-27:
@MaddTheSane  do u have a link to these cables?

closed 2018-03-29 by FranticRain