360Controller/360Controller #604: GameSir g3s wireless  working on Sierra OSX 10.12.6 (but two controller issue)

sadarific, 2018-02-16:
Let me save you some time. I was able to play with the GameSir g3s controller on Sierra OSX 10.12.6 using the xbox 360 drivers: 360Controller 0.16.5 https://github.com/360Controller/360Controller/releases . Find the instructions below:

1.-Install drivers.
2.-Restart.
3.-System preferences ->Xbox 360 controllers->Advanced-> Check "Enable Driver".
4.-Plug USB dongle.
5.-Power ON the controller. (You will see the red dot blinking which indicates it hasn't paired yet)
6.-Try different combinations (Home + Y, Home + X) in 1 minute intervals until the red dot stops blinking (it indicates that it has been paired) and you will see appear the device in the Xbox 360 controllers window. Make sure you configure the controller in the X-input mode (Home + X), it didn't work for me if you configure it in the mac mode ( Home + Y).
7.- DO NOT DO ANYTHING IN THE WINDOW.
8.-If you do, then you will probably have to reboot your system by powering off your mac on the hard way (pushing the on/off button for 5 seconds) due it won't open programs like steam and you will not be able to turn off your system in the safe mode. 
9.-In any case, you will be able to use your controller now, never open the "Xbox 360 Controllers" window again. Just plug your USB dongle (always plug it on the same USB port, or repeat the instructions if you want to enable another one) and have fun. 

After all these boring instructions, please do not waste your time trying to connect another controller. It just don't work. Trust me, I did a lot of trials with several combinations, installing and uninstalling, restarting my mac several times to figure this out. 

I hope someone could fix all this issues with the XBOX 360 drivers for mac (Sierra OSX 10.12.6). I would give him a nice hug. 

I am not sure if a real XBOX 360 controller would work using these instructions. 

Greetings,

FranticRain, 2018-04-02:
This controller is not a real Xbox 360 controller and is not supported by this driver. We wont fix, and we won't support controllers of this type. This is not an issue.

closed 2018-04-02 by FranticRain

Luqrecio, 2019-01-16:
The driver works fine. I've just played a racing game.
Thanks!

ime-01, 2023-04-05:
I cannot connect it for some reason. I've lost my dongle and so I am using a wired connection. Even if I connected my controller via Bluetooth, it would work perfectly fine, but it disconnects in no more than 3-4 seconds. With the wired connection, it works fine too, but only on an online gamepad tester https://gamepad-tester.com/ and no other app recognizes the controller, not even this 360Controller driver.

I don't know how to fix this issue, so can someone help me out here?