360Controller/360Controller #605: WARNING: DO NOT ENABLE “Drew Mills” on High Sierra 10.13.3.

ulyssesric, 2018-02-18:
I tried to install ver 0.16.5 on High Sierra 10.13.3, but it couldn’t recognize my wired XBox 360 controller. Restart system with controller connected did’t work either. Some posts here suggested that turning on “Drew Mills” in Security & Privacy would fix the problem. After I did that, system asked for permission and then rebooted, and it just failed to boot again.

The progress bar will fill up but never complete, even after 30 minutes. Due to some unknown reason, I **can’t even boot into Safe Mode**. Resetting NVRAM and SMC doesn’t work either. In the end I reinstalled the whole system from Recovery Mode. Thanks god all my data and settings still remained intact (my last TM backup was 7 days ago).

**DO NOT INSTALL ver 0.16.5 on High Sierra 10.13.3.**

baldengineer, 2018-02-19:
I'm running 10.13.3 but did not have this issue.

pauliharman, 2018-02-28:
I didn't have that problem, but my wired Xbox controllers (I have several of different vintages) would not pick up.

JSuttiIsNutti, 2018-03-02:
He’s completely correct. I just enabled Drew Mills and my Mac won’t start back up. Progress bar is stuck. Please fix this.... I need to use my machine. I am seriously regretting downloading this.

JSuttiIsNutti, 2018-03-02:
Alright, so 2 hours later, I found a solution.

-Turn off your Mac machine
-press power button and immediately hold down the command and "R" keys simultaneously
------this will boot the machine into recovery mode.
-connect to a wireless network
-reinstall the OS
-go to system preferences > xbox 360 controller > advanced and click uninstall

This sucked. 

Is there any way to ensure that the developer is not still enabled on my machine?

MaddTheSane, 2018-03-02:
I just removed the wireless driver.

Part of the problem is that the driver needs to be updated to work with new APIs introduced with El Capitan.

JSuttiIsNutti, 2018-03-02:
....Someone else do it first and let me know how it goes. 

But in all honesty, thanks for the quick response and solution. I'll try it again when ulysseric or someone else tries it. This machine is brand new, don't want to risk it again.

JSuttiIsNutti, 2018-03-03:
I thought there would be a reply by now, but there isn't and I'm itching to try this software out.

MaddTheSane: are you saying that the problem is fixed? Or are you saying it was only part of a larger issue?

MaddTheSane, 2018-03-03:
The issue is not fixed, but seems to be limited to the wireless connector/driver.

JSuttiIsNutti, 2018-03-03:
Alright. I was trying to use the wired connection and the software wouldn't recognize my controller even though I could find the controller in a USB-C slot in the system report. So I'll wait here until the software is usable.

d235j, 2018-03-03:
Has anyone attempted to do two-machine (or VM based) debugging to see where the panics are happening? I would but don't have time and would need to locate my wireless adapter (not sure where it is).

goodtimeclub, 2018-03-04:
Same symptoms here, @ulyssesric  – posting specs & basic rundown of what happened below:

**Steps:**
1. installed latest .dmg release, 16.5 (Logitech F310 controller not connected)
2. allowed developer access under Security/Privacy
3. finished installation, reboot initiated automatically
4. progress bar hit ~70% but never passed that – followed that with a hard reboot
5. same thing
6. booted into Safe mode - progress bar hit 100% but never completed
7. booted into Single User mode, mounted root filesystem as rw
8. moved /System/Library/Extensions/360Controller.kext to my Desktop (couldn't find any other .kext's named "Extensions/Wireless*.kext as per [this thread](https://github.com/360Controller/360Controller/issues/525))
9. rebooted & everything is as it was before

**Specs:**
- late 2009 iMac running High Sierra 10.13.3
- 14GB ram
- Logitech F310 controller (never plugged in but I do have Logitech's "Control Center" installed)
- black office chair with hydraulic lift feature

I'll try uninstalling this release & give it a try later with a controller plugged in the whole time, maybe that matters. Will report back tomorrow – good luck to anyone else suffering from a stuck boot!

goodtimeclub, 2018-03-05:
Checked my work laptop this morning & it didn't experience the same problem. I installed the same way on Friday (no controller attached) and it rebooted fine. Only differences here are the hardware and that my home iMac has the Logitech Control Center installed. Will report back once I know more...

joeymoss, 2018-03-08:
Created an account just for this issue. I had the same the same issue, stuck at 70% followed by hard reset. My iMac now will not recognize any bluetooth keyboards. It will boot to the login screen but will not recognize any keyboard so I am locked out of my machine. Mouse works fine. Anyone with any idea how to fix?

d235j, 2018-03-10:
@MaddTheSane @FranticRain can we put out a release with the wireless driver removed as a temporary workaround to mitigate stuck boot? (I don't have the ability to sign kexts.)

FranticRain, 2018-03-11:
I think I'm the only one with a kext certificate, but I updated to High Sierra. Can't run Xcode 6.4 anymore to do the builds. That's pretty much why development was halted. We could simply release the El Cap version where the wireless 360 controller stuff was never completed. Thoughts?

MaddTheSane, 2018-03-11:
My El Cap branch isn't working.

FranticRain, 2018-03-11:
Well then there isn't a whole lot that can be done, I guess.

d235j, 2018-03-11:
I will look into methods for compiling against the appropriate SDK with current Xcode / macOS and post back here.

d235j, 2018-03-13:
@MaddTheSane @FranticRain 

Grab [XcodeLegacy](https://github.com/devernay/xcodelegacy)
Then run
`./XcodeLegacy.sh -osx1010 buildpackages`
(Following the instructions to download the appropriate version of Xcode and place it in the right location)
`sudo ./XcodeLegacy.sh -osx1010 install`

That loads the correct SDK into Xcode and make them usable for building the .kext. It is necessary to select the 10.10 SDK in the project.

I don't remember — did we compile with llvm-gcc or clang and against which stdlib?

nacnewmedia, 2018-03-13:
Welp. This escalated quickly 😐

MaddTheSane, 2018-03-13:
I remember the kexts needing to be compiled using llvm-gcc.

d235j, 2018-03-13:
@MaddTheSane thanks, will test and update instructions 🙂.

d235j, 2018-03-13:
@MaddTheSane it looks like compiling with Clang is only a problem when targeting the 32-bit kernel... which was removed in 10.8 anyway so should not be an issue. Just need the 10.10 SDK, for which the above instructions work.

FranticRain, 2018-03-13:
Someone will have to check my work. It's been a little while. Had to make it a zip because github.
[360ControllerInstall.dmg.zip](https://github.com/360Controller/360Controller/files/1808354/360ControllerInstall.dmg.zip)

codewisp, 2018-03-21:
Tried installing the one from the zip and it made my system startup hang. It might just be taking a really long time (doing a disk check or something, because it's the same in Safe Mode, just took longer to fill up the white bar), so I'll report back.

codewisp, 2018-03-21:
The release version worked better, it seems :)

FranticRain, 2018-03-21:
Well. That's the install that doesn't include the wireless kexts, so I'll need some more guidance if this is going to get fixed.

c7borg, 2018-03-22:
Note you do NOT need to reinstall the OS!

An extra note for how to go into write mode as I found googling can lead you to some time consuming methods.

Again the same symptoms as @ulyssesric and @goodtimeclub  

Steps:

installed latest .dmg release, 16.5 (Rock candy XBOX 360 controller not connected)
allowed developer access under Security/Privacy
finished installation, reboot initiated automatically
progress bar hit ~70% but never passed that – followed that with a hard reboot
same thing
Wouldn't boot into safe mode by holding down shift
Booted into Single User mode using cmd & s
Just above the prompt it will tell you how to mounted root filesystem so you can delete/move the file
On mine it was:
/sbin/fsck -fy
/sbin/mount -uw /
Then move /System/Library/Extensions/360Controller.kext to some other place like the Desktop or for me I just deleted it
rebooted & everything is as it was before

Specs:
MacBook Pro(Retina 13inch, Mid 2014) running High Sierra 10.13.3
8GB ram
Rock Candy Xbox 360 controller wired

FranticRain, 2018-03-22:
You had to remove /System/Library/Extensions/360Controller.kext? Interesting. Here's another build that might fix the problem, then.

[360ControllerInstall.dmg.zip](https://github.com/360Controller/360Controller/files/1838029/360ControllerInstall.dmg.zip)

c7borg, 2018-03-22:
Thanks, but oddly it all worked fine after removing it - quite happly
playing fornite... I've added that to my post as not to put people off.

Good work though - thanks for the update.

FranticRain, 2018-03-29:
I still would appreciate it if someone who is willing to potentially reinstall their OS is willing to give that version a shot to see if it fixes the broken boot situation.

DigitalRama, 2018-04-04:
I gave that version a bash just now it I was able to reboot, shutdown, boot without any issue. I did this while making sure I unplugged the USB connecting my Xbox One Controller and unticking Enable Driver in 'Advanced'.

hl, 2018-04-05:
I'm running MacOS 10.13.3 and after installing 360Controller 0.16.6 my Macbook Pro (15'' 2017) I rebooted.

Next thing that came up was install recovery screen from MacOS telling me that MacOS failed to install. Reboot or shutdown resulted in the same loop. I did manage to get out of it by just selecting reboot in the menu on top instead of the reboot button shown in the popup warning.

hl, 2018-04-05:
For what it is worth, I did not see the 'Drew Mills' message in the pref panel nor did my controllers work afterwards

Slick3gz, 2018-04-09:
MacBook Pro Mid 2014 
macOS High Sierra 10.13.4

Hung on last release. Used zip from Frantic’s last post. Rebooted fine. Controller working.

FranticRain, 2018-04-09:
@Slick3gz Last release, as in 0.16.6 that was released less than a week ago? Because 0.16.6 and this zip are identical. I assume you mean 0.16.5.

Slick3gz, 2018-04-09:
@FranticRain Sorry for the confusion. My Mac hung after installing 0.16.5. I booted into single user mode, mounted my drive and removed the extensions in Library/Extensions. Was able to get back to the desktop and install the zip from your post. Rebooted without issue and was able to use my original Xbox One controller plugged in with a micro USB cable.

evandegr, 2018-05-11:
Hi I’ve just tried to install this and am crashing on boot. I don’t see the 360controller.kext stuff. Is there another way to fix this outside re installing the OS?

vergen, 2018-07-05:
I am having the same issue with 10.13.5 and 0.16.8. I am able to start my computer but I'm getting errors about bluetooth keyboards, security issues with bluetooth keyboards and it's also tellling me that I have to press shift + the key closest to the right of shift to identify this mysterious keyboard I don't have connected.

My install is always stuck at 75% every time I try to install the drivers. The kextload command you specified in another thread doesn't give me any output.

It also gives me an error when uninstalling so I'm not really sure what is actually installing and what is not.

I've allowed your certificate after the security prompt in the installer.

windytan, 2018-08-01:
I had the same problem with 0.16.8 and macOS 10.13.6:
* After installing the driver, my system didn't boot and the progress bar got stuck at 75% as described in this thread.
* I removed the kext alias file /System/Library/Extensions/360Controller.kext in single user mode (Cmd+S) and the system booted up, but controller did not work.
* I installed the version from @FranticRain 's .zip (marked 0.16.5 there). System booted up, but I realized I have an unsupported third-party controller anyway and I'll file another issue about that.

mtraps, 2018-08-02:
I want to remove all the drivers I've recently installed and start from scratch. How would I go about doing so?

FranticRain, 2018-08-06:
@mtraps You use the uninstall feature built in to the driver to remove all of components. They all install over each other and the uninstaller will remove all components from every variation of the driver ever installed.

If you want to do it by hand, here are the paths to all of the elements you will need to remove.
```
/Library/LaunchDaemons/com.mice.360Daemon.plist
/Library/Application Support/MICE/360Daemon.app
/System/Library/Extensions/360Controller.kext
/System/Library/Extensions/Wireless360Controller.kext
/System/Library/Extensions/WirelessGamingReceiver.kext
/Library/Extensions/360Controller.kext
/Library/Extensions/Wireless360Controller.kext
/Library/Extensions/WirelessGamingReceiver.kext
/Library/Extensions/XboxOneBluetooth.kext
/Library/PreferencePanes/Pref360Control.prefPane
```

Furthermore, please keep your posts to your own issue instead of posting on unrelated issues. This only clutters and confuses issues, hurting the likelihood that the original poster will have their problem resolved.

mtraps, 2018-08-06:
Gotcha sorry about that, and thank you.

swiperthefox93, 2018-08-22:
Hi guys, same problem, I read everything to a “T” and I must be typing the commands in single user mode incorrectly can anyone provide more of walkthrough in exactly how I’m supposed to input /sbin/fsck -fy
/sbin/mount -uw /

When I typed the first one it told me it didn’t recognize the “-f” part and also said it would choose to ignore it I’m assuming that’s not what we want 

Assistance would be highly appreciated as I spend several hours a day on this laptop and all I wanted to do was play fortnite with a controller and now I’m starin at a wall

FranticRain, 2018-08-23:
@swiperthefox93 If you are using a wired Xbox 360 or Xbox One controller, try the "Pretend to be a 360 Controller" option in the "Advanced" section of the preference pane. If you are using a wireless controller, that is an issue with Fortnite. Furthermore, your issue is unrelated to this issue entirely, and I please ask you to make your own separate issue instead of pinging a thread of people who have an unrelated problem.

swiperthefox93, 2018-08-23:
My apologies, my comment was initially to confirm that your method worked. I was learning the errors at the same time and figured “while I’m here” but I already figured it out last night and yes spoofing a 360 controller also did work. With that said there’s no need for another thread and thank you for your help :)

salsaseltzer, 2019-03-14:
I tried installing today and had the same issue, had to re-install Mojave and luckily that did it, still have no Xbox controller driver though.

neutroh, 2019-06-08:
This issue is not fixed on the latest v1.0.0 alpha 3. Boot hangs on Mojave. DO NOT INSTALL these drivers unless you want to reinstall your OS.

shdnx, 2019-07-13:
Same issue for me with the latest released version (1.0.0-alpha.3 from 2019 May 13) with macOS Mojave (version 10.14.5) running on a 2017 MacBook Pro. After installation and restarting, macOS hangs on login. Removing `360Controller.kext` in recovery mode solved the issue.