360Controller/360Controller #606: Add support for Xbox 360 Scene it controllers

joan, 2018-02-18:
Hi!

Would be awesome to support the Scene It big buttons:

[image]

Seems to works on PC using https://github.com/sverrirs/XboxBigButton

Any chance to see it arrives on a Mac ? 🙏

arthas994, 2018-02-21:
You could simple use it by mapping the buttons, give it a try.

joan, 2018-02-21:
I wish I could!

But it's not recognized by macOS otherwise than as an “Vendor specific device (by Microsoft Corporation)”.

Linux and Windows both need drivers (xbox360bb / XboxBigButton), only then it's mappable. But no driver seems to exist for macOS, and neither 360Controller nor Enjoy2 / Enjoyable / Joystick Mapper / USB Overdrive recognize the device.

Yes, I gave it a several days try — and finally ended with an Ubuntu VM…

Maybe you have something in mind I missed? 🙏

arthas994, 2018-02-24:
Hi!
You can try to add the product ID (you can find it in the same page where you found the vendor id) to plist file and then map the controller..
I’m not sure it is gonna work but you can give a try

joan, 2018-02-26:
Hi,

Thanks for highlighting fact that we can add devices! So I was starting to add these controlles… until I saw they are already in the plist file!

https://github.com/360Controller/360Controller/blob/master/360Controller/Info.plist#L1574
Vendor/product ID are the same (0x045e → 1118 / 0x02a0 → 672).

So it means that… it just does not work?

joan, 2018-02-26:
So, when I connect the IR receiver, I have almost a hundred of `IOUSBHostDevice: family specific matching fails` in system console, followed by `Xbox360Peripheral::probe(IOUSBHostDevice)` and `Xbox360Peripheral::start(IOUSBHostDevice) <1>`.
Normal behaviour? 🤔

Cruplezone, 2018-03-03:
the wireless adapter for wireless controllers code is not maintained that well 😬not sure if that applies to this particular ir receiver too though.

saying that just to justify it's not your fault or you doing anything wrong. probably just been a bug for a while.

joan, 2018-03-03:
Thanks to clarify, then! Any chance to see this code updated? 🙏

Cruplezone, 2018-03-03:
Neither Frantic nor Madd have expressed interest in maintaining the wireless receiver code. once again I'm not sure if that is related to this specific receiver or not, but it's in the same field at least.

In fact Madd commented a couple of days ago he has removed wireless driver altogether due to its incompatibility with yosemite apis. Not sure if its merged into master yet.

In otherwise, looks bleak for wireless accessories 😣

joan, 2018-03-03:
Owh, what a shame! I was definitely hoping to see it work on macOS and stop relying on an Ubuntu VM 😔

Cruplezone, 2018-03-03:
Maybe tag them and express your interest in such development? I know they've tried before and analyzing the wireless protocol is difficult but it's possible. Least you can do is show interest

joan, 2018-03-04:
Let's send good vibes to @FranticRain and @MaddTheSane then!  
Having these buttons working could allow me to make great games in group 😍

FranticRain, 2018-04-02:
Going to be hard to implement. We'd need an understanding of the packet layout of this controller, which is going to be pretty hard, unless those things work in Windows. If they do, then try to sniff the packets using the same method as we use for Xbox One controllers. Without an understanding of the packet layout, we can't make it work. https://franticrain.github.io/sniffing/index.html

Cruplezone, 2018-04-03:
@FranticRain does this mean you will be actively deving for this once again? I see you filtered through many issues recently. 

Slightly related, is there a chance the kernel panics with sleeping while wireless 360 adapter is plugged in be looked into? Do you need more sniffing done for that device?

Thanks for coming back and even offering to help even with these odd controllers

FranticRain, 2018-04-03:
Yes and no. I'm looking to do some cleanup related to a potential fix for the wireless 360 controller situation. Which is to say that we will be removing the wireless 360 controller kext from a coming release. I still don't know how the wireless stuff works, so I'm not sure I could ever fix it. Mostly I'm just looking to accumulate some new devices and a couple odd fixes for weird issues, like the preferences fix. Maybe shore up some stuff related to new kext guidelines. The wireless controller stuff won't be fixed with more scanning. It just needs to be torn up and written from scratch using the new IOUSBHost guidelines. I have a heck of a time justifying that hellish time investment though.

I'd like to get the Big Buttons in there if it isn't too much work. I think your bigger problem is really going to be with games supporting them. Once again, I have no idea how the wireless stuff works. I don't know how multiple controllers match through a single wireless receiver and all sorts of stuff. I can give it a shot, but I promise you absolutely nothing.

electric-monk, 2018-04-03:
The Big Buttons thing is unrelated to XBox 360 wired and wireless support. Long ago someone sent me one to figure out support, and the protocol is pretty simple, but I never was able to actually support it and now I am prevented from doing so for various reasons. The dongle just appears as a device that reports an enormous array of buttons, from what I recall, and there's no pairing as they're basically just infrared remote controls, so it'd have to be its own unrelated driver anyway.

Regarding wireless 360 support, one thing I'd been intending to do was to refactor it to work with the same code as the wired support. I originally did it the way I did (with separate loadable drivers for controllers/etc.) to support things like the steering wheel, but since I never got my hands on one it was all just an exercise in futility anyway. I got started on that refactoring but again, some things prevent me right now.

Cruplezone, 2018-04-03:
Hey no one expects anything from you guys so I wouldn't worry about that. It's crazy how far this simple driver has come from its original to even absorb other projects.

At this point it seems with the advent of BT in the newest controllers (and steam directly interfacing with them with their own proprietary driver) this will be a legacy controller driver that won't have much to add in the future. Thankfully.

Anyways a removal of the wireless driver seems justified for now. It would be very cool to see it get official support in the future (next couple of years in guessing as a time line), but this is volunteer coding at work. Community thanks you!

joan, 2018-04-03:
Many thanks for coming back on this and still trying to keep these alive!

> unless those things work in Windows

No idea if this can help but as I said, both [Linux](https://github.com/micolous/xbox360bb) and [Windows](https://github.com/sverrirs/XboxBigButton) drivers exist (at least I tried the first one on Ubuntu via a VM, works fine).
I had a hard time to find the controllers but finally got them so I'll be happy to provide any information about them if needed (still got the VM working).

Joan, 2019-09-15:
Some interesting updates about this driver: any plan on supporting these big buttons? 🙏