360Controller/360Controller #607: Xbox one controller not connecting to computer

whatevermydudes, 2018-02-20:
I have installed the program and it seems that it is set up correctly, but it wont show up. The controller is plugged in correctly but neither the controller or the computer is detecting each other. I'm running on version 10.11.2 if that changes anything.

arthas994, 2018-02-21:
If you are able to see your controller in system report you should try to allow Drew Milis developer under Security & Privacy in System preferences.

Kelvne, 2018-03-17:
@arthas994
Thanks! It worked for me.

sethhaney4, 2018-03-25:
That does not show up for me

FranticRain, 2018-03-29:
https://github.com/360Controller/360Controller#xbox-one-controllers-connected-with-usb

FranticRain, 2018-04-02:
If the computer can't see the device, then it is likely that you are using what is know as a "charge" cable. These cables do not provide data to and from the controller, therefore the driver cannot do anything with it. Try a different cable.

FranticRain, 2018-05-17:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-05-17 by FranticRain