360Controller/360Controller #610: Update the driver to follow new best practices for driver loading

doctorpangloss, 2018-02-24:
I was updating WJoy, Bluetooth HID driver for Nintendo drivers, and I encountered the same issue that your users are reporting regarding no tips on what to "do next." The problem seems to occur on High Sierra.

There's an Apple Tech Note that describes the new changes: https://developer.apple.com/library/content/technotes/tn2459/_index.html

I would suggest that instead of using `kextload` in:
https://github.com/360Controller/360Controller/blob/0a6990c894a5c5e45c631a1f1b72bb30475dc5dc/Pref360Control/Pref360ControlPref.m#L1064

Try to load the driver directly with `KEXTManager`:
```
    CFStringRef bundleId_ = (__bridge CFStringRef) @"com.mice.driver.Xbox360Controller";
    OSReturn result = KextManagerLoadKextWithIdentifier(bundleId_, NULL);
    if (result == kOSKextReturnSystemPolicy) {
        // Prompt the user to enable the driver in Security & Privacy
    }
```

FranticRain, 2019-05-30:
I've been looking into fully implementing this enhancement, and I have to say that I'm not entirely sold on the concept. In order for an application to load the kernel extension, it has to be operating as a superuser. That means that using `KextManagerLoadKextWithIdentifier` in the preference pane it out. So the ask is: Establish a connection from the preference pane to the daemon so that we can use its elevated privileges to load a kernel extension.

Or I could just prompt the user for their password and run the terminal commands from AppleScript.

The choice seems pretty obvious. If I'm missing some legitimate reason to actually put loading into the daemon, let me know. But that just sounds like a heck of a security concern.

doctorpangloss, 2019-06-03:
Two notes:

 - The daemon stuff is just from old Apple documentation, I don't really know if it makes sense.
 - There's a subtle interaction with preferences panes and the Security & Privacy > Privacy > Accessibility setting. If apps are running that are allowed to control your computer, which is true on many gamer's computers (Discord asks for this permission, for example, and Steam used to), sometimes security buttons cannot be clicked on. This is [undocumented](https://apple.stackexchange.com/questions/323668/allow-button-not-working-in-system-preferences) and the specific behavior changes with every OS version.

Using the most official way of doing things probably interacts with this problem the most favorably. At the time, this accessibility thing was the root cause of people not seeing what other random people on the Internet said the user should be prompted to do w.r.t installing drivers. So it's all very subtle and frustrating.

FranticRain, 2019-06-03:
I've been digging into this ever since that post and I've found that the most effective way to get error codes out of `kextload`/`kextutil` appears to be through the daemon, so I've been implementing it following the daemon practice. I don't believe that this best practice change would solve the approval issues though. At best, I can at least give them a prompt that says that they need to check the box. But the hidden block won't be solved by doing it this way. I currently see this issue as implementing a better error/warning system when a user clicks the "enable" checkbox that can help the user debug the issue without needing to open a GitHub issue. But I don't think I can do much more than that.

Are there any other action items you are suggesting?

FranticRain, 2019-10-19:
This should be fixed in alpha 4.

closed 2019-10-19 by FranticRain