360Controller/360Controller #615: xbox

mperaz21, 2018-03-14:
so hey. I am an xbox one player that owns a mac and i really need help. i want to connect my xbox one controller to my mac but i have been getting many problems. i downloaded the latest version of Frantic Rain's 360 driver but when i plugged in my controller via Usb it only vibrated and didn't light up. then, i deleted my old xbox one file, the one that got closed, and it still doesn't light up. i checked my hardware and it recognizes the controller but it still doesn't light up. i am on mac os high sierra. Please help!!!!

FranticRain, 2018-04-02:
https://github.com/360Controller/360Controller#wired-xbox-360-controllers

You say that the hardware recognizes it. I assume that you mean that the hardware information shows the controller. Does it show up in the preference pane settings? What are the vendor and product IDs?

FranticRain, 2018-05-17:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-05-17 by FranticRain