360Controller/360Controller #621: Xbox 360 Controller Not Registering

hayntal, 2018-03-22:
Hey,

I have downloaded and installed v0.16.5, and uninstalled and reinstalled, a few times to try and get my wired 360 controller to register on a new MacBook. It works perfectly on my older computer.
I have a data enabled usb converter plugged in, and when I go to use the driver it says "no devices found". Any idea?

tundo91, 2018-03-27:
Having same problem with a Wireless 360 controller on High Sierra and Sierra. I tried to multiple versions: 0.16.5, 0.16.4, 0.16.3, 0.16.2, 0.16, 0.15 beta 6, nothing good.
I don't even see "Allow" button in "Security & Privacy". Any idea?

FranticRain, 2018-03-29:
Wireless Xbox 360 controllers are supported with the Xbox 360 Wireless Gaming Receiver. Using a "Play and Charge" kit is not supported.

The thing you are using does not make a wireless controller wired. It simply charges the controller.

closed 2018-03-29 by FranticRain

tundo91, 2018-03-29:
I'm actually using a 360 wireless controller + a Wireless Gaming Receiver like [this one](http://cronusmax.com/manual/images/drex_xbox_360_wireless_controller_custom.png). This should be supported right?

FranticRain, 2018-03-29:
Yes, but that's not the issue that the OP was having. Wireless stuff is out of my domain. If you open up an issue, you might get some help. It's likely that you have a third party knock-off receiver, and those are notoriously incompatible with the driver.

hayntal, 2018-03-29:
You’re confused, I’m using a wired controller. Not wireless. 

- Hayden Antal

FranticRain, 2018-03-29:
Did you try any of the debugging tips in the README? https://github.com/360Controller/360Controller#wired-xbox-360-controllers