360Controller/360Controller #622: Hori Fighting Commander not working

Nitro-Oxide, 2018-03-24:
My Hori Fighting Commander (Xbox edition) is not reacting to the driver nor is the driver picking the controller up. The controller itself will not turn on, even when plugged in. However, the mac itself is able to detect the controller
I am on macOS sierra (10.12.6), I

Product ID: 0x00c5
Vendor ID: 0x0f0d

FranticRain, 2018-06-01:
Could you give the latest release a shot and let me know if it works? There was a change in init packets that might have resolved this issues.

FranticRain, 2018-08-12:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-08-12 by FranticRain

Nitro-Oxide, 2018-08-23:
This might be too late, but I am happy to report the driver is now working
for my gamepad.
Thank you