360Controller/360Controller #624: Xbox One controller isnt recognized on mac high sierra 10.13.3

lilphil123, 2018-03-27:
ive tried 3 different cables and it isnt working.
can anyone send me a link to a cable that'll work for sure? idk if thats the problem though

no controller pops up.
the xbox light just blinks when i plug in usb and may vibrate.

i have the old xbox controller

lilphil123, 2018-03-27:
it isnt the cable because my controller shows up in the system repot usb hardware settings.

lilphil123, 2018-03-27:
@FranticRain ?

FranticRain, 2018-03-29:
https://github.com/360Controller/360Controller#xbox-one-controllers-connected-with-usb

FranticRain, 2018-05-17:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-05-17 by FranticRain