360Controller/360Controller #625: Intermittent "Bluetooth not available" on High Sierra

jollino, 2018-03-27:
Hello,
I installed 360Controller for use with OneCast and everything's fine with 360Controller itself (I use an older Xbox One controller, the kind without the 3.5mm audio jack), but ever since installing it I've had some intermittent issues with Bluetooth.

I am on a 2017 iMac 27" 5K (iMac18,3), i.e. the latest model at this time, on High Sierra 10.13.3 and Bluetooth randomly goes from working perfectly to saying "Not available", as if the adapter is physically disconnected. This started after installing 360Controller for the first time, it had never happened before in the six months I had had this machine.
I have already tried the usual mumbo jumbo — clearing nvram, deleting preferences, rebooting in safe mode to clear caches, even removing 360Controller — but the issue seems to persist. I've basically tried everything short of reinstalling OS X from scratch.
It usually happens when coming out of sleep, but can happen at any random time. Rebooting always fixes it so I'm inclined to think that it's not a hardware issue (one never knows, though!)

Note that I'm not saying it's necessarily because of 360Controller, I was just wondering if there's any chance that there may be some odd incompatibility with newer Macs that may persist even after uninstalling 360Controller.

Thanks in advance!

FranticRain, 2018-05-17:
It is quite possible that there are lingering kexts that were not removed by the uninstaller. If you could check `/Library/Extensions` for kexts, especially ones like `XboxOneBluetooth.kext` and see if you see anything with `BL` or `Bluetooth` in the name that might have been left behind, that would be outstanding. You may have to use the `CMD + Shift + G` shortcut in finder to gain access to this directory.

jollino, 2018-05-18:
I have nothing of the sort in there, however in the meantime I have pretty much found that it seems to be unrelated — there must be some sort of odd USB conflict in High Sierra on my specific configuration as I can more or less replicate it reliably: attaching new USB devices (hard drives especially) seems to kill Bluetooth. Disconnecting them and using blueutil (found on homebrew) restarts it. Oddly enough, this happens only after the machine has been put to sleep multiple times. On a fresh boot, it's stable. Sorry for not updating this before, it had completely escaped my mind as I haven't had much time to play to begin with!

`ytsejam:/Library/Extensions jollino$ ls -l
total 0
drwxr-xr-x  3 root  wheel  96 Jun 13  2014 ACS6x.kext
drwxr-xr-x  3 root  wheel  96 Jun 27  2016 ATTOCelerityFC8.kext
drwxr-xr-x  3 root  wheel  96 Jun 27  2016 ATTOExpressSASHBA2.kext
drwxr-xr-x  3 root  wheel  96 Jun 27  2016 ATTOExpressSASRAID2.kext
drwxr-xr-x  3 root  wheel  96 Aug 20  2013 ArcMSR.kext
drwxr-xr-x  3 root  wheel  96 Sep  1  2013 CalDigitHDProDrv.kext
drwxr-xr-x  3 root  wheel  96 Aug 15  2014 HighPointIOP.kext
drwxr-xr-x  3 root  wheel  96 Aug 15  2014 HighPointRR.kext
drwxr-xr-x  3 root  wheel  96 Mar 31  2017 PromiseSTEX.kext
drwxr-xr-x  3 root  wheel  96 Aug 22  2017 SoftRAID.kext
`