360Controller/360Controller #626: XBOX360 racing wheel for Mac

Cybershoot, 2018-03-28:
Hi, I have a microsoft wireless controller and it works very well on Sierra.
So I want to try the receiver with my XBOX360 wireless racing wheel.
It connects and the lights come on, but the commands do not respond, he doesn't work.
Where I can find the drivers for Sierra?
Or in the future will they implement them for the steering wheel?

Thank You.

FranticRain, 2018-04-02:
https://github.com/360Controller/360Controller#wired-xbox-360-controllers

Does it show up in the preference pane? What are the vendor and product IDs?

Cybershoot, 2018-04-02:
@FranticRain  Yes, it show. Its connects to the wireless. If I switch ON/OFF the steering wheel the pane turn ON or OFF. But no control, no button is respond in the steering wheel. It doesn't work.

[image]

I use the original Microsoft 9Z2-00002 Wireless Windows PC.
Xbox 360 Controller Driver ver. 0.16.5

[image]

MacBookPro 15-inch Mid 2014 - MacOS Sierra ver. 10.12.6 
2.5 GHz Intel Core i7
16GB RAM - 1600 MHz
Graphic card Intel Iris Pro 1500

FranticRain, 2018-04-02:
Wireless controllers are not officially supported in Sierra and later. More so with racing wheels. There is not wireless controller programmer around anymore, so chances of getting this fixed are very low. It's likely that it is a different packet layout than the controller layout, which is causing the problem. You'd have to get help from someone like @electric-monk

Cybershoot, 2018-04-03:
I've tried with xbox 360 wireless **controller pad** it works very well, and the wired controller pad is good also. The problem is the xbox wireless racing wheel, it doesn't work. Probably just for what you said.

Cybershoot, 2018-04-04:
Unfortunately I'm not a developer. I have read that the xbox steering wheel uses "Xinput api".
as from the site. 
[http://www.tomshardware.co.uk/answers/id-2247641/microsoft-xbox-360-racing-wheel-xbox-360-wireless-adaptor-speed-run.html](url)
Until now there is no solution. 
I hope it can help someone, or to give some ideas in the future.
I could be a beta tester. :)