360Controller/360Controller #629: Fully Wired (No wireless feature) Third Party (PDP) XBOX One Controller Not Found

ThePriceIsWrong, 2018-04-01:
I have a third party fully (exclusively) wired xbox one controller I have been trying to connect. I just downloaded and installed the 0.16.4 release and I only get "no devices found" under my preferences pane. I have never had the old Xone-OSX drivers or any other xbox driver,  I also attempted to delete and re-download the 0.16.4 release. Help or at-least give me a wakeup call that this just "ain't gonna work", it would be much appreciated. I am running OSX 10.13.3 by the way.

FranticRain, 2018-04-02:
https://github.com/360Controller/360Controller#xbox-one-controllers-connected-with-usb

FranticRain, 2018-05-17:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-05-17 by FranticRain