360Controller/360Controller #631: Xbox One Controller Not Recognised

Jaymon27, 2018-04-02:
When I connect my Xbox controller to my MacBook Air, the Xbox button on the controller lights up but it doesn't get recognised by the driver. It works on my other mac though, both of which are running the same OS and the same version of the driver. It's strange it's not working because it was fine yesterday but today it just decided it didn't want to pick up... @FranticRain

FranticRain, 2018-04-02:
Are both computers running the same version of macOS? 10.13.3 just came out. Is it possible that one of them updated?

mturbin6688, 2018-04-02:
My computer is running 10.12.6 and the original xbox one controller (its a COD AW color way controller) was working just fine. Then, randomly, the thing just stopped working. My computer still recognizes the controller in the system report, but the 360 controller no longer recognizes my controller. What can I do to fix it? Again, I had been using it in the same Mac OS version for quite some time before that- and it randomly stopped working. Ive uninstalled and reinstalled 360 controller, and it still isn't working. Any advice you have would be greatly appreciated. @FranticRain

FranticRain, 2018-04-02:
I would appreciate it if you made a separate issue for your separate problem, but it's too late now. I need your PID/VID off of your controller, just to make sure. Follow the instructions here to find it: https://github.com/360Controller/360Controller/blob/master/Readme.md#xbox-one-controllers-connected-with-usb

Additionally, does your controller light up in any fashion?

mturbin6688, 2018-04-03:
I will get the PID/VID for you later- I don't have my controller with me. However, the controller fully lights up (steady, not flashing) and the controller itself still lights up in the same exact way it did back when the driver was working on my Mac. I've used the driver to play fortnite on my Mac before, but now the driver no longer registers the controller, despite the controller's complete recognition by the computer. I'm hopeful that I can, with your help, get it to work again.

mturbin6688, 2018-04-03:
@FranticRain my controller's product ID is "0x02d1" and my controller's vendor ID is "0x045e (Microsoft Corporation)" Just to be clear- I added the quotes. How might this controller have been affected?

mturbin6688, 2018-04-03:
Wait- I just plugged in an xbox one S controller to see if that would work. Surprisingly, it recognized the newer, bluetooth compatible (albeit still wired) S controller, but now no longer recognizes the older controller. The New controller is "0x02ea" and "0x045e". Any explanation? @FranticRain

Jaymon27, 2018-04-04:
@FranticRain the computer that is working is using the version 13.2, whereas my laptop that isn't working is using 13.3.

FranticRain, 2018-04-04:
@mturbin6688 Did you use the exact same cable with both controllers?

@Jaymon27 I'd be inclined to believe that if it just stopped working and you just installed a new version of your operating system, then that would likely be the problem. Is it possible that you need to allow the driver in System Preferences -> Security and Privacy -> General, again?

[image]

mturbin6688, 2018-04-05:
Same Exact cable @FranticRain . Whats interesting is that in looking in the system report, the incompatible controller is running version 1.01, and the working controller is running version 3.01

Jaymon27, 2018-04-05:
@FranticRain 
Nope, I've definitely accepted that alert. I'm hesitant to update my other mac to 10.13.3 in fear it will stop working on that too!

FranticRain, 2018-04-06:
@mturbin6688 Update your controller using either an Xbox console or a Windows 10 PC.

@Jaymon27 My best guess then is that 10.3.3 breaks the driver for good. I don't know what could possibly be going wrong. You'll need to use `Console.app` in `Applications/Utilities` and open up the `system.log.` From a fresh boot, plug the controller and see if anything new shows up. I don't need your whole log, just the new stuff.

FranticRain, 2018-04-06:
@Jaymon27 Might as well update the one machine from 10.13.3 to 10.13.4 just to see if it magically fixes itself.

FranticRain, 2018-06-16:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-06-16 by FranticRain