360Controller/360Controller #633: Can't Connect Second Wired XBOX One Controller

meegaaann7, 2018-04-03:
I connected an Xbox one Controller to my mac. I am now trying to connect a different controller but it will not recognize it as a new controller. No buttons light up when it is connect with 360controller downloaded. I have uninstalled and reinstalled 360controller but the same problem happens. Please Help.

meegaaann7, 2018-04-03:
Additional Information: My controller when plugged in is listed in system overview as a Controller under USB. This is an XBOX one controller directly from XBOX not a third party controller.

FranticRain, 2018-05-31:
So, just to be clear, both of these are official Microsoft Xbox One controllers? Can you plug them in one at a time and retrieve this vendor and product ID's for these controllers via this method: https://github.com/360Controller/360Controller#how-do-i-find-my-vendor-id-and-product-id

FranticRain, 2018-06-23:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-06-23 by FranticRain