360Controller/360Controller #634: Xbox wireless controller triggers not working

Lechucico, 2018-04-04:
Hello,

If my controller is connected via USB, the Xbox 360 Controller driver detects it and everything works as expected. Works perfect on games.

If my controller is connected via Bluetooth, both R and L triggers doesn't work and my d-pad doesn't work. Works perfect on steam big picture but won't work on Brawlhalla. I have to plug it to make it work again.

Any idea how to fix it?

FranticRain, 2018-04-11:
The Bluetooth controller is outside the scope of this project, and furthermore your issue is game specific. So this issue is doubly outside the scope of support. Here's the standard line we give on game specific issues, where the same is true of the actions of the Bluetooth controller:

The driver does not provide any sort of compatibility for individual games beyond what is provided in the preference pane. The driver simply sends button data to the operating system. Once it does that, control is out of the driver's hands. Since Apple does not provide any standardized means of interacting with controllers in macOS, every game uses a different method to read controller data. We cannot provide individual fixes for each game within the controller. That simply isn't feasible. It is the responsibility of game developers to support controllers, not the other way around. We cannot and will not provide support for individual games. If you are using an Xbox One controller, we can recommend that you use the "Pretend to be an Xbox 360 Controller" option in the "Advanced" menu for better game compatibility. Beyond that, there is nothing we can do.

closed 2018-04-11 by FranticRain

xXH4tchXx, 2018-07-11:
This isn't game specific, the mappings of the trigger axis are different when registered through bluetooth. It should be possible to catch this input and correctly map the axis' and d-pad. I reached out to the gamepadtool guys for help on this, we'll see what they can add. 

This thread has some helpful input feedback towards that bottom that I believe will enable you guys to fully support the bluetooth xbox one controller.https://steamcommunity.com/app/400450/discussions/0/133259227529975635/

Lastly, the release notes and description of this plug in are misleading, it implies the xbox one wireless controller(connected by bluetooth) is fully supported but, that isn't the case.

FranticRain, 2018-07-12:
Okay, you're right. It isn't game specific. It's a rebinding issue. Considering this driver doesn't support the Xbox One S controller in Bluetooth mode, it's out of scope. Additionally, rebinding, especially of triggers, is also considered out of scope.

Now, let's talk about the idea that we should "fix" this binding. The current bindings are technically wrong. The way the controller binds to values via the driver is quite literally wrong, following the HID spec. So, given a controller that properly fits the HID spec, the works with the operating system without a driver, I think making a driver for the controller that changes the binding would be absolutely disastrous. Suddenly a single controller supports multiple layouts based on whether or not you have a driver installed. I refuse to be responsible for that.

But the conclusion, or rather the punchline, of this issue is that it's an open source repository. If you feel that the documentation is misleading, you're more than welcome to correct it and make it more clear. If you feel that there should be a driver for the Bluetooth controller to fix that binding, you're more than welcome to make it. Be my guest. But personally, knowing what I know about how macOS works with games and controllers, I wouldn't touch that with a 10 foot pole.