360Controller/360Controller #635: Just try 0.16.6 dmg say disk image cannot be opened?

Chacapamac, 2018-04-05:
Just try 0.16.6 dmg say disk image cannot be opened?
Mac OS ElCapitan 10.11.6

Errors message when double click dmg is “**no mountable file system**”

AlexBezuska, 2018-04-06:
The same issue happened to me, the previous release worked. 
Mac OS Sierra 10.12

Chacapamac, 2018-04-11:
Thanks Alex !
Version Version 0.16.5 work perfectly on ElCapitan with a third party wired x-box style controller. To play on Steam, I tick the “**Pretend to be an Xbox Controller**” in the “**Advanced**” tab of the Mac preference. I Was able to also register the controller on Steam, no problems.

Great Stuff! - Bravo for the Dev.

FranticRain, 2018-05-17:
Latest version that adds the new devices should work on systems earlier than 10.13.

EricFromCanada, 2018-05-17:
Afraid not… v0.16.7's disk image is still APFS.

FranticRain, 2018-05-17:
Hmm. It should have had the fix from the PR. I'll look at it again.

FranticRain, 2018-05-17:
Okay. I think I found the problem. I changed the dmg on the release page. Please let me know if it works.

Chacapamac, 2018-06-05:
Yes work perfectly! Thanks…

commit 11b6c71b (2018-05-17): Force makedmg.sh to HFS+
Should actually fix #635