360Controller/360Controller #636: Stuck Reboot

Spudanky, 2018-04-07:
I’ve installed the driver and restarted my iMac, but now stuck at 50 percent at the apple logo loading screen. I re-restarted in safe mode, and that got me farther in the progress bar, now it’s stuck at around 75 percent.

squizzeak, 2018-04-12:
I experienced the same issue under 0.16.6.  Removing and reinstalling 0.16.5 resolves the issue.

  System Version:	macOS 10.13.4 (17E199)
  Kernel Version:	Darwin 17.5.0

Tek022, 2018-04-19:
Can confirm 0.16.6 hangs 10.13.4.  Needed to use Recovery to remove the .kext's in order to boot.

MacBookPro14,3
macOS 10.13.4 (17E199)
Darwin 17.5.0

daschk, 2018-05-07:
Also confirming 0.16.6 hangs 10.13.4. Removing and reinstalling 0.16.5 resolves the issue.

Bhoritz, 2018-05-09:
How can you erase just the .kext from disk utility in recovery? 
(And yes, same problem with 0.16.6, the one you are directed to use because stuck reboot is fixed in it)

daschk, 2018-05-10:
Restore to a time machine checkpoint. Or start in single user mode.

Tek022, 2018-05-10:
@Bhoritz You can open up a Terminal inside the RecoveryOS and remove the installed .kexts as well, which is how I did it.  They'll be in `/System/Library/Extensions`, I don't remember the exact names, but if you run `ls -al | grep -i xbox` while inside that folder you should find them.

Bhoritz, 2018-05-10:
Thanks. Using the terminal did work. Not sure if I'll install 10.16.5. It was the one supposed to stuck your Mac after all.

Cluisanna, 2018-07-09:
I’m having the exact same issue except that I can find no file with xbox or 360controller in the name in /System/Library/Extensions. Both grep and reading through the entire ls doesn’t reveal anything helpful. I’m an absolute noob and only made a GitHub account to ask for help here because I have no idea what else to do except restore my month-old-backup from time machine, which I would prefer not to do because it would mean losing some work.

daschk, 2018-07-09:
See here:  [https://github.com/360Controller/360Controller/issues/657](https://github.com/360Controller/360Controller/issues/657)