360Controller/360Controller #638: No devices found

AntB95, 2018-04-08:
Hi, 
I just installed Xbox 360 Controller on my computer but when I'm connecting my controller : "No devices found"
I know that my controller works on PC and I'm using a MacBook Pro 13' on MacOS High Sierra 10.13.4 

Thanks for your help

Antoine

onlinewithryan, 2018-04-09:
Same issue here. I have never installed 360 controller on this machine before.

Updated to MacOS High Sierra 10.13.4, then downloaded and installed 360 controller.
No controllers were recognized so I dove into the settings.

In the settings I constantly see "no devices found"

Going into advanced tab, under options I see that "enable driver" is unchecked. No matter what I do to check it, it will not save that state.

[image]

[image]

tommo, 2018-04-10:
Check the 'Security and Privacy' panel in preferences, the driver might be forbidden to run by system there.

FranticRain, 2018-04-11:
Before we can help you, please consult the README and provide all necessary information so that we can help you with your issue. https://github.com/360Controller/360Controller#wired-xbox-360-controllers

FranticRain, 2018-04-23:
@AnasHaw From the documentation:
> Any problems with game compatibility in Bluetooth mode are completely out of our control and are up to you to solve in conjunction with the game developer.

For the wired portion, make a new issue, I assume you are posting here because your preference pane also says "No devices found." Does your controller appear in the "About This Mac" USB report? What is it's product and vendor number? Did you check the security and privacy panel as mentioned earlier in this post?

mahdaen, 2018-05-08:
My Controller not detected on the pref pane too. For reference, below the USB info:

```
Controller:

  Product ID:	0x02a0
  Vendor ID:	0x0738  (Mad Catz, Inc.)
  Version:	1.00
  Serial Number:	3A8D92F8
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Microsoft Corporation.
  Location ID:	0x14d20000 / 15
  Current Available (mA):	500
  Extra Operating Current (mA):	0
```

alonsoluis85, 2018-05-08:
Hello everyone!

I have the same issue with my controller. I have found a not very practical or safe work-around. I don't really think it is a viable solution but it might help you figure our how to solve this problem with Mac OS 10.13.4 (at least that's the one I have and the controller doesn't work with usb...).

Searching a solution for another issue on the Steam forums I found this thread:
https://steamcommunity.com/app/353370/discussions/1/1484359403778266904/

A user named Mitchismo posted this solution to their problem (not being able to install some Steam controller drivers):

1) Reboot into Recovery Mode (Command-R at Reboot)
2) Disable SIP (Enter "csrutil disable" in terminal in Recovery Mode)
3) Reboot
4) Run Steam & install the component (successfully)
5) Quit, Reboot into Recovery mode (Command-R at Reboot)
6) Enable SIP (Enter "csrutil enable" in Terminal)
7) Reboot and Enjoy your updated controller.

I thought there could be a similar issue for the xbox 360 driver and it seems like it is! As long as the SIP is disabled you can properly install and use the driver (the controller works again) and as soon as you enable it and reboot the controller stops working. It seems like it is a security issue within Mac OS? I don't really understand any of it but it works :-)

I hope this can help somehow. Good luck and thank you for your awesome work! @FranticRain

FranticRain, 2018-05-09:
This process is well known and documented in the README. https://github.com/360Controller/360Controller#adding-third-party-controllers

I don't know what your problem was or why this would help at all in solving it. It is probably tied to not accepting the driver in System Preferences -> Security and Privacy -> General.

[image]

alonsoluis85, 2018-05-09:
Thank you for the quick answer @FranticRain !
Unfortunately during the installation process I never had the option to accept the driver in system preferences. I tried to install it with Mac OS 10.13.4.
I am going to try to install everything again tonight see if I did anything wrong. 
Have a great day!

FranticRain, 2018-06-01:
This issue involves too many individual issues and seems to have been either abandoned or resolved by its respective participants. Please open a new issue if you still have one.

closed 2018-06-01 by FranticRain

AnimeShrek568, 2019-12-14:
> This process is well known and documented in the README. https://github.com/360Controller/360Controller#adding-third-party-controllers
>
> I don't know what your problem was or why this would help at [...]

I have the same problem. I did this before, Allowed the drivers but it still doesn't work

jbarros35, 2019-12-22:
I don't even have Steam, its about that Drew Mills stuff?

khanvict, 2021-04-06:
I am having the same exact issue, were you able so find a solution?

twentythrill, 2022-06-07:
> Hello everyone!
>
> I have the same issue with my controller. I have found a not very practical or safe work-around. I don't really think it is a viable solution but it might help you figure our how to [...]

Just tried this by running both "sudo spctl --master-disable" and "csrutil disable" from recovery but didn't help.
The driver looks like has been installed correctly but I still get a "no devices found" when plugging in the controller.
None of the games nor emulators I've installed on my machine recognise the controller but weirdly enough https://gamepad-tester.com/ does, it also recognise every button and stick input correctly, despite this I still can't figure out how to make it work on the rest of the OS.