360Controller/360Controller #640: Any new for Home-brew cask

prenfic, 2018-04-11:
```
Hello, im just new to this thru the homebrew cask new tap,
and wonder,
is there any news or new for the xbox one slim controller with audio jack,
or,
is there some1 ever test or try please tell me that,
is it working for this,
for the MacOS 10.13.4.
```

Regards,
kjL

FranticRain, 2018-04-11:
1. The homebrew cask is not officially supported, as it isn't signed.

2. The controller you're talking about has been supported for a long time. In fact, macOS supports it in Bluetooth mode natively. Use it that way. Do not use this driver.

3. It isn't officially supported in macOS 10.13. Some people have success, others have had to reinstall their OS.

prenfic, 2018-04-11:
@FranticRain Man, thx for your these comment, since, im still working at Taipei Time, 02:57 although, thus, I havent reboot or shut my system down, you may save me, thx ya!!!

prenfic, 2018-04-11:
I must say, you just save me from a nightmare maybe.

[image]