360Controller/360Controller #647: 0.16.6 Speedlink Torid vibration no longer works

Diamondog36, 2018-04-22:
After updating to 0.16.6, vibration no longer woks in games, at least with this gamepad. While testing it in the driver preferences, I noticed than it doesn't vibrate when pressing RT and LT, as it should. Emulators don't even recognize the rumble feature of the gamepad. At first I thought it was a hardware issue, but I tested it on Windows and it works fine. I don't have another gamepad to test, so, I don't know if a change broke some gamepads, or all of them. It is USB-wireless, so, it is considered wired.

FranticRain, 2018-04-23:
Hmm. Probably happened because of my lazy build to get 0.16.6 out the door, but I don't know why that would have happened. Sorry. I'll look into it if I get some time.

FranticRain, 2018-06-01:
This should be fixed in the latest release.

closed 2018-06-01 by FranticRain

joobei, 2019-01-08:
My torid gamepad isn't even recognized. @Diamondog36 can you please confirm if Speedlink Torid works?

Diamondog36, 2019-01-08:
Yes, it works for me. I'm using 0.16.10. This is an old thread by the way it was closed months ago, I was asking about an older version. You should create a new issue about this, not resurrect older issues.

> My torid gamepad isn't even recognized. @Diamondog36 can you please confirm if Speedlink Torid works? It's not listed under supported devices in the wiki.

joobei, 2019-01-09:
> This is an old thread by the way, it was closed months ago, I was asking about an older version. You should create a new issue about this, not resurrect older issues.

I cannot open a new issue for a controller that is not in the list of officially supported controllers. That is why I asked you here.