360Controller/360Controller #649: No Device Found

Tbails, 2018-04-23:
Hi,
Really appreciate the effort gone in to put this together.

Unfortunately, I cannot get the controller to work on my MacBook pro.

Every time I try and connect it the controller vibrates for a second but no light comes on. 

I have read the HELPME file and still no luck. Tried using multiple cables and controllers as well. 

Any help would be appreciated.

FranticRain, 2018-04-23:
Before we can help you, please consult the README and provide all necessary information so that we can help you with your issue. https://github.com/360Controller/360Controller#xbox-one-controllers-connected-with-usb

Tbails, 2018-04-30:
Hi, i even ready the READMR However I am unclear as to exactly what it is you need in terms of info?

FranticRain, 2018-05-01:
I need you to provide the "Product ID" and "Vendor ID" of the controller. These can be found by accessing the Apple menu, selecting "About this Mac", and then selecting "System Report..." on the "Overview" tab. On the left hand side of the new window, select the "USB" option under "Hardware". If the controller is plugged in, there should be an entry in this window called "Controller".

Tbails, 2018-05-02:
It does not appear, when plugged in, on this screen. The xbox logo does not light up on the controller either it only vibrates for a moment.

FranticRain, 2018-05-03:
> If it isn't listed there, then you likely have a "charge" Micro USB cable instead of a "data" cable. If the cable isn't sending data, then you can't use the driver. Try a different cable.
From the README.

Tbails, 2018-05-14:
Hi again,

I now have a cable which works and have the following information
Product ID:	0x02d1
  Vendor ID:	0x045e  (Microsoft Corporation)

FranticRain, 2018-05-14:
Please ensure that the driver has been allowed in System Preferences -> Security and Privacy -> General. This approval UI is only present in the Security & Privacy preferences pane for 30 minutes after the alert. Until you approve the kext, future load attempts will cause the approval UI to reappear but will not trigger another user alert.

[image]

Tbails, 2018-05-14:
Right, but I have not had this software UI, is there any way to trigger it or check if I have allowed it before?

FranticRain, 2018-05-14:
I'm not sure there's any way to check if you have allowed it already, but the way to prompt this UI is to connect the controller after the driver has been installed. This UI will only show up in 10.13.4 and later, I think. I forget when exactly it was implemented. 

I guess the best way to check if you have approved it is to run `sudo kextstat | grep com.mice ` in your terminal fairly soon after connecting the controller. If the result is empty, then the driver hasn't loaded. If you see an entry, then the driver has been properly loaded.

Tbails, 2018-05-14:
Does not look like there is anything 
Last login: Mon May 14 18:38:46 on ttys000
Thomass-MacBook-Pro:~ thomasbailey$ sudo kextstat | grep com.mice
Password:
Thomass-MacBook-Pro:~ thomasbailey$

FranticRain, 2018-05-14:
Can you try `sudo kextload /Library/Extensions/360Controller.kext` and tell me what the output is?

FranticRain, 2018-06-16:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-06-16 by FranticRain