360Controller/360Controller #651: URGENT - Stuck on startup, please help

Kapleo, 2018-04-26:
Hi,
I downloaded this hoping to have my xbox 360 controller work, instead it's completely stopped my computer from working - it's stuck on loading (boot up) after logging in to a user.
I've read the other issues threads and attempted some fixes, but none are very clear on exact instructions (I'm not code savvy)
Is anyone able to help me please? Thanks so much

bachmanity1289, 2018-06-04:
How did you fix this without losing all of your computer data? This thing ruined my computer too

FranticRain, 2018-06-05:
@Kapleo I'm sorry, but that's actually impossible. I've put warnings up all over this piece of software that say that it's dangerous as heck to install a kext on your machine. I can install the driver and use the controller *perfectly* on my system. I do this work for free, I have one Mac that I only use for development of this driver, and I don't even play games on macOS. If you'd like to personally test every release before I upload it, please feel free to get into contact with me. But I **DID** test this release, and it works. As to what the cause is that broke on your system, I don't know. I'd love to fix it, but it's a bug in macOS that I can't fix. Because if you go back to earlier versions of macOS, this doesn't happen.

If you'd like, I can just stop updating the driver, because I simply can't test that this thing works perfectly on everyone's configuration before I release the dang thing.

Closing because it's resolved, second poster has his own issue that contains a response that resolves the issue.

closed 2018-06-05 by FranticRain