360Controller/360Controller #652: PDP Wired controller for Xbox one - no device found

bitfinity, 2018-04-27:
Installed twice - privacy and security settings etc. I'm a software and devops guy.

[image]

[image]

bitfinity, 2018-04-27:
Tried to compile latest Xcode as well but the USB header file no longer the same.

MaddTheSane, 2018-04-27:
The changed headers is a known issue (#127). It will take a re-write of the drivers to get it building using newer Mac OS X SDKs.

closed 2018-05-01 by FranticRain