360Controller/360Controller #655: NEED HELP! Xbox One Controller Freaks Out When Connected Via Bluetooth!

JordanKigour, 2018-04-27:
I have a Xbox One Controller trying to connect via bluetooth to my 2017 Macbook Pro. It has difficulty finding the controller and when it does finally connect the controller freaks out and acts as though something is holding down the buttons constantly spamming them. It also displays a grey circle with a line through it in between the jumps.

When this occurs, the trackpad will work but it will keep skipping back to the spot it was at and doesn't allow me to exit a game, window or anything! Very frustrating. 

Don't know how to solve this issue.

I have tried also using the Xbox 360 Controller Driver and it will allow me to select "enable driver" but the "Pretend to be an xbox 360 controller" is greyed out and cannot check the box.

I don't mind if I use it wired or wireless, I just want to find out what i can do to fix one of the above issues. 

Much appreciated!
- Jordan

FranticRain, 2018-05-01:
This driver does not interact with the Bluetooth version of the driver. There was an attempt to add force feedback to the controller, but all of the interactions between the Bluetooth controller and the operating system are outside the scope of this driver. It cannot help you with these issues. Nothing in the driver will impact the operation of the controller in Bluetooth mode. There is no support for the wireless protocol with the Xbox One controller adapter. The protocol has not been cracked, so we cannot implement it. The driver only provides support for Xbox One controllers connected to the PC via a USB connection.

closed 2018-05-01 by FranticRain

JordanKigour, 2018-05-01:
Thank you very much for the response.