360Controller/360Controller #658: Mac BookPro V:10.13.3 Wont recognise the xbox remote

IMALAPACA, 2018-05-03:
hi I installed the latest version of the 360controller today on my MacBook Pro v-10.13.3, and it doesn't seem to recognise my xbox one remote. I have used the original cable for the remote then I tried my Samsung and then some other generic brand but still couldn't get a response. I even uninstalled and reinstalled it multiple time with and without the remote in the computer. can u help me pls?

FranticRain, 2018-05-04:
Before we can help you, please consult the README and provide all necessary information so that we can help you with your issue. https://github.com/360Controller/360Controller#xbox-one-controllers-connected-with-usb

In this case, we need the vendor and product IDs as specified in the README.

FranticRain, 2018-06-16:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-06-16 by FranticRain