360Controller/360Controller #659: Controller thumbsticks don't reach edge - allow calibration?

HaniiPuppy, 2018-05-03:
https://drive.google.com/file/d/1gAjAfrF5x6QEpB_GC_UUN_xCXXZVsm7H/view?usp=sharing

I have the PDP Legendary Deliverer of Truth controller. The thumbsticks don't seem to reach the edge of the area, especially the corners. Obviously, this isn't an issue of the driver, but rather with the device itself. However, it could be dealt with in the driver generically, by allowing calibration by having the user roll the thumbstick in a circle like that, tracking the top, bottom, left, and right sides of the usable area, then using that information to establish a diagonal, then using that to track the four corners, then scaling the input coördinate accordingly.

It wouldn't just deal with this controller's quirks specifically, but any that has a seemingly non-standard mapping for the thumbsticks and possibly even some faulty controllers.

FranticRain, 2018-05-04:
I forget exactly how the deadzones work, it's been a while, but does the "normalize" option provide any relief for this issue? I don't think it does, unless you have an interior deadzone.

HaniiPuppy, 2018-05-05:
Deadzones aren't the issue, it's that the controller's input doesn't reach the edges, even when the thumbstick is pushed as far as it can go.

FranticRain, 2018-05-08:
The "normalize" option is designed to modify the range of the stick. If you view the location of the stick as a value between 0 and 1, then the normalize option takes a stick input with a dead zone of 0.2, and remaps this 0.2-1 input range to 0-1. The idea being here, that perhaps it solves your issue. I fully understood your issue, and would appreciate if you would do as I asked. The potential fix for this would be an outer dead zone. Outer dead zones are outside the scope of the driver at this time. If you or someone else would like to program and support such a feature, I would be happy to add it via a pull request.