360Controller/360Controller #663: Spoofing Preferences: Pretend to be an xbox 360 controller

Jakedjoe, 2018-05-05:
Small but annoying issue:

under Spoofing Preferences, Pretend to be an xbox 360 controller toggle button is not saving the preference. My third party PowerA controller works fine when this preference is toggled. However, when I close the setting menu the preference does not save. So I need to launch the menu every time I want to play, toggle the button, and keep it open; or else the mapping for the game is completely off.

FranticRain, 2018-05-09:
This should have been fixed in 0.16.6. What version are you using?

Jakedjoe, 2018-05-10:
version 0.16.6

FranticRain, 2018-06-01:
This should be fixed in the latest release.

closed 2018-06-01 by FranticRain

wbbradley, 2020-04-26:
This issue (or perhaps a variant of it) still persists. When I close the lid on my MacBook Pro, and reopen it, the spoofing preference seems to relapse to Off, when I had set it to On. Is this a known issue?