360Controller/360Controller #665: NO DEVICE FOUND

ericmuller02, 2018-05-07:
I have just downloaded 360Controller and when I go to plug in my controller, the light will blink and it will show up as no device found. I have already tried 2 cables. What am I doing wrong?

daschk, 2018-05-07:
Most cables are "charge only" cables. I had to try 3 cables and the third one worked. 

See [https://github.com/360Controller/360Controller#xbox-one-controllers-connected-with-usb.](https://github.com/360Controller/360Controller#xbox-one-controllers-connected-with-usb.)

If the controller does not appear as a USB device try a new cable.

ericmuller02, 2018-05-07:
I'll try that, thanks

FranticRain, 2018-06-16:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-06-16 by FranticRain