360Controller/360Controller #667: Support for Razer Wolverine Ultimate

martineriksson, 2018-05-08:
I have tried getting 360Controller to work with the Razer Wolverine Ultimate but despite many attempts I have not managed to build the driver after adding a new entry to Info.plist. I keep running into problem after problem related to running Xcode 6.4 on my up-to-date Mac.

I would be excited if someone who is able to build the project could fix this. Here is the relevant details:

Product ID: 2580
Vendor ID: 5426

Please let me know if I can help in some other way! If someone else can build the driver and provide the resulting .pkg file, I will be more than happy to test it and report back.

Here is the entry I added to the Info.plist file:

```
		<key>RazerWolverineUltimate</key>
		<dict>
			<key>CFBundleIdentifier</key>
			<string>com.mice.driver.Xbox360Controller</string>
			<key>IOCFPlugInTypes</key>
			<dict>
				<key>F4545CE5-BF5B-11D6-A4BB-0003933E3E3E</key>
				<string>360Controller.kext/Contents/PlugIns/Feedback360.plugin</string>
			</dict>
			<key>IOClass</key>
			<string>Xbox360Peripheral</string>
			<key>IOKitDebug</key>
			<integer>65535</integer>
			<key>IOProviderClass</key>
			<string>IOUSBDevice</string>
			<key>idProduct</key>
			<integer>2580</integer>
			<key>idVendor</key>
			<integer>5426</integer>
		</dict>
```

closed 2018-05-16 by FranticRain

FranticRain, 2018-05-17:
This device is in the latest build.

martineriksson, 2018-05-17:
Thanks a lot!

I can confirm that the Razer Wolverine Ultimate works with the latest build.

seanhealystudio, 2020-08-26:
I have the latest version of the program installed, but the wolverine ultimate still doesn't seem to be registering inputs for me. @martineriksson , how were you able to make it work?

mateofelipe, 2020-08-26:
Make sure your Xbox isn’t near or it’ll connect and void the pc connection. This happens a lot for us. You can plug directly into the pc or Mac and it’ll power up the controller without you having to push the power button. This will avoid communication with the Xbox. It’s been working great with this approach for us.

seanhealystudio, 2020-08-26:
I don't have an Xbox, so that isn't the problem. The issue for me seems to be that 360controller detects that the thing is plugged in, but doesn't identify any inputs. I'm plugging the controller into the laptop directly, via a micro-usb to USB c cable (I've also tried using the original cable, as well as several others).

mateofelipe, 2020-08-26:
Ok. I also made sure to download the version listed much earlier in the comments. I found it helpful to read through them from the top. There was an elite controller customization added to a version incorporated into one of these early posts and that is the version I have. Should be clear in the comments. Sorry I’m not more helpful. I do still find it to be glitchy sometimes but usually a reboot solves the issue.

Good luck.

martineriksson, 2020-08-27:
> I have the latest version of the program installed, but the wolverine ultimate still doesn't seem to be registering inputs for me. @martineriksson , how were you able to make it work?

I have not done anything special, I simply installed the driver and it worked right away. It still works for me and the version I am using at the moment is 1.0.0-alpha.2 and I am on macOS 10.14.6. Hope this helps!

seanhealystudio, 2020-08-27:
Bizarre – I’m also using the newest version of 360controller , though I’m on macOS 10.15.5. I suppose my OS could be the issue? Maybe it doesn’t work on Catalina? I’ve gotten other xbox controllers to work, so this seems extremely odd…

martineriksson, 2020-08-27:
I am _not_ using the latest version. Read my last comment again if you want to try doing what I am doing.

seanhealystudio, 2020-08-27:
Just tried your version, no luck. I guess it's Catalina? Otherwise, I'm at a loss.

commit c353fead (2018-05-16): Add backlog of devices
Resolves #675
Resolves #667
Resolves #662
Resolves #656