360Controller/360Controller #668: Xbox Controller saying devices not found

xboxdaddy89, 2018-05-11:
Hi, I am having a problem with my driver. It seems whenever I plug in my controller it will not light up nor vibrate. I checked system report and it is detecting that my controller is plugged in! please help1

FranticRain, 2018-05-13:
What type of controller is it? Is it Xbox One, 360, original Xbox? What brand is it? What is the vendor/product ID of the controller?

FranticRain, 2018-06-16:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-06-16 by FranticRain