360Controller/360Controller #669: No devices found

makobreako, 2018-05-11:
I tried plugging in my xbox one controller to my mac book pro (with latest updates) and it does not recognize my controller.  I did look at the other people with the same issues, so I also tried using a bunch of other micro usb cables, and all failed.  Not sure what to do now.

FranticRain, 2018-05-13:
Before we can help you, please consult the README and provide all necessary information so that we can help you with your issue. In your case, we need the vendor and product ID as specified in the README. https://github.com/360Controller/360Controller#xbox-one-controllers-connected-with-usb

FranticRain, 2018-06-16:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-08-12 by FranticRain