360Controller/360Controller #671: No Devices found

maximS1, 2018-05-13:
Inno GamePad..:

  Product ID:	0x028e
  Vendor ID:	0x045e  (Microsoft Corporation)

Unsure why it wont work I did everything that was required, would be great if you could help, thanks!

maximS1, 2018-05-13:
Also this is a 360 controller and not an xbox one controller

FranticRain, 2018-05-13:
Please ensure that the driver has been allowed in System Preferences -> Security and Privacy -> General. This approval UI is only present in the Security & Privacy preferences pane for 30 minutes after the alert. Until you approve the kext, future load attempts will cause the approval UI to reappear but will not trigger another user alert.

[image]

It's also entirely possible that, because this is a counterfeit controller, it just won't work with this driver correctly. There is nothing we can do about that, sorry.

FranticRain, 2018-06-23:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-06-23 by FranticRain