360Controller/360Controller #674: Xbox 360 wireless controller 'No devices found'

riceandpixel, 2018-05-15:
Hi. I have tried to set up my Xbox 360 Wireless controller to work with my iMac. I am running macOS High Sierra 10.13.4 and have installed version 0.16.4 of the XBox 360 Controller Driver (because it said in the notes for 0.16.5 that support for wireless controllers had been removed) 

The wireless controller is connected to an Xbox 360 Wireless Receiver for Windows: (I have used this device to successfully connect my wireless 360 controller to an iMac in the past) - I've had a look at the system information report and I think it's showing the info for the Wireless Receiver , not the controller?

  Product ID:	0x0719
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	1.00
  Serial Number:	FD27DCC0
  Speed:	Up to 12 Mb/sec
  Manufacturer:	©Microsoft
  Location ID:	0x14400000 / 2
  Current Available (mA):	500
  Extra Operating Current (mA):	0

I have been to Security and Privacy in System preferences and hit the 'Allow' button to approve "Drew Mills" as a developer. I did notice that the 'Approve' button and message has not disappeared so am not sure if the command had been accepted by the system or not. 

I read "No Device Found #649"  (https://github.com/360Controller/360Controller/issues/649) and saw the terminal commands you suggested - I have tried using 'sudo kextstat | grep com.mice' to see if the driver has loaded but I don't get any response (See below)

27IMAC2:~ javied.idrees$ sudo kextstat | grep com.mice
Password:
27IMAC2:~ javied.idrees$

I have also tried using 'sudo kextload /Library/Extensions/360Controller.kext' and i get the following response: 

27IMAC2:~ javied.idrees$ sudo kextload /Library/Extensions/360Controller.kext
/Library/Extensions/360Controller.kext failed to load - (libkern/kext) system policy prevents loading; check the system/kernel logs for errors or try kextutil(8).
27IMAC2:~ javied.idrees$ 

Any help or advice you can offer is appreciated. 

Thanks

FranticRain, 2018-05-15:
So that is the error that is thrown when the "approve" value isn't correctly accepted. I'm not quite sure what your issue is with that, but I know that there have been issues for users that have software that alters mouse control or are using screen sharing. Or for example, users with Wacom tablets can't accept the dialog with their tablet, they have to do it with a physical mouse. So I think the solution to this problem would be to ensure that you have properly accepted the alert in system preferences, so that the driver can load.

But I will warn you that what you are doing is very much not a great idea and you should proceed with caution. The reason the wireless controller support was removed was because it causes a kernel panic in 10.13. If you still want to use it, make sure to power off your controller and disable the driver in the preference pane before your computer goes to sleep or shuts down. Otherwise, your operating system WILL kernel panic. It was a bug introduced by Apple in a compatibility layer for their new driver system that we can't fix, and that they refuse to fix.

FranticRain, 2018-05-15:
Yes! Wired Xbox 360 controllers should be fully featured. Just a heads up here as well that the "play and charge kit" that would seemingly turn a wireless controller into a wired one doesn't actually do that. It's just a means of charging the controller. It does not transfer data. You'll need a controller that has a wire by default. The whole "adding a wire to a wireless controller to make it wired" thing wasn't a feature until the Xbox One controller.

I'm going to go ahead and close this down. Feel free to reply if you have any further questions.

closed 2018-05-15 by FranticRain