360Controller/360Controller #677: Macbook Pro Will not recognize

lpvinyl21, 2018-05-17:
After installing the latest Xbox 360 Controller Driver today, my MacBook Pro 10.13.4 will not recognize my controller which was working fine before. I even tried installing the previous version and still the Xbox 360 Controllers says no remote detected.

At this point, I have reinstalled and rebooted my mac a few times. I y even checked to see if I needed to "allow it" under the security & privacy settings and there's nothing there. 

Any help would be appreciated! 

Controller:

  Product ID:	0x02ea
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	3.01
  Serial Number:	3032363030313632313536373239
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Microsoft
  Location ID:	0x14300000 / 1
  Current Available (mA):	500
  Extra Operating Current (mA):	0

FranticRain, 2018-05-17:
With the controller plugged in, please run sudo kextload /Library/Extensions/360Controller.kext in your terminal. And let me know what the output says.

lpvinyl21, 2018-05-18:
Hi FranticRain, When I tried running this, it prompted me for a password. 

Last login: Thu May 17 22:32:39 on console
Lindas-MBP:~ lindaterry$ sudo kextload /Library/Extensions/360Controller.kext
Password:
Sorry, try again.
Password:
Sorry, try again.
Password:

FranticRain, 2018-05-18:
It should be the same password that you use to log in to your computer. It is simply prompting you to provide administrator control to the command. Don't worry, the driver doesn't get to see any of this. This is all for the purposes of Apple's own kextload tool.

lpvinyl21, 2018-05-18:
I thought that might be the case but for some reason, it doesn't let me type there. I'm not sure what I'm doing wrong.

FranticRain, 2018-05-18:
Does the computer have multiple accounts? You need to be signed in to an administrator account that has a password in order to use `sudo`. I'm not sure why it would fail otherwise. If you go to `System Preferences` -> `Security & Privacy` and click the lock, it should prompt you for a password. Does your password work there?

lpvinyl21, 2018-05-18:
No, I'm the only account. I googled it and figured out how to input the password. 

Here is the result from the command prompt. Thanks!

Password:
/Library/Extensions/360Controller.kext failed to load - (libkern/kext) authentication failure (file ownership/permissions); check the system/kernel logs for errors or try kextutil(8).
Lindas-MBP:~ lindaterry$

FranticRain, 2018-05-18:
Interesting. This is very strange. It's almost as if the driver was installed without administrator privilege. We can try and correct this by running: `sudo chown -R root:wheel /Library/Extensions/360Controller.kext`. So do that, then try to run `kextload` again and let me know what it tells you.

lpvinyl21, 2018-05-18:
That is strange because I'm the only one who uses this computer. 

here are the results, nothing happens when I run the second command you gave me.

Thank you so much your help.

PLast login: Thu May 17 23:39:53 on ttys000
Lindas-MBP:~ lindaterry$ sudo chown -R root:wheel /Library/Extensions/360Controller.kext
Password:
Lindas-MBP:~ lindaterry$ sudo chown -R root:wheel /Library/Extensions/360Controller.kext
Lindas-MBP:~ lindaterry$ sudo kextload /Library/Extensions/360Controller.kext
/Library/Extensions/360Controller.kext failed to load - (libkern/kext) authentication failure (file ownership/permissions); check the system/kernel logs for errors or try kextutil(8).
Lindas-MBP:~ lindaterry$

FranticRain, 2018-05-18:
`chown` shouldn't tell you anything, it's just changing the ownership of some files. Basically, it's doing what the error suggested. My best guess is that `sudo kextutil 360Controller.kext` returns something like this:
```
Notice: 360Controller.kext has debug properties set.
Kext rejected due to improper filesystem permissions: <OSKext 0x7f8cd1807890 [0x7fff9046caf0]> { URL = "file:///Library/StagedExtensions/Users/drewmills/Desktop/360Controller/build/Release/360Controller.kext/", ID = "com.mice.driver.Xbox360Controller" }
Kext rejected due to improper filesystem permissions: <OSKext 0x7f8cd1807890 [0x7fff9046caf0]> { URL = "file:///Library/StagedExtensions/Users/drewmills/Desktop/360Controller/build/Release/360Controller.kext/", ID = "com.mice.driver.Xbox360Controller" }
Diagnostics for 360Controller.kext:
Authentication Failures: 
    File owner/permissions are incorrect (must be root:wheel, nonwritable by group/other): 
        /Library/StagedExtensions/Users/drewmills/Desktop/360Controller/build/Release/360Controller.kext
        Contents
        _CodeSignature
        CodeResources
        MacOS
        360Controller
        PlugIns
        Resources
        English.lproj
        InfoPlist.strings
        Info.plist

Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents
```

I have absolutely no idea why you are seeing this. I can actually recreate this issue by simply building the project through Xcode and not using the installer provided in the releases section. The issue is fixed as soon as I move the file to `/Library/Extensions` though. I'm sorry, I just don't know how to fix this. It seems like something at the OS level is broken and I don't know how to figure out what it is.

lpvinyl21, 2018-05-18:
Oh no :( Well thank you so much for your help.

lpvinyl21, 2018-05-18:
Just one more question if that's ok. Would reinstalling my mac OS help with this you think? Thanks!

FranticRain, 2018-05-18:
Possibly. I can't recommend it as an option, but if you were to do a full clean installation then it might fix this. Something did occur to me while working on the driver earlier, though. Is your controller plugged in during the installation process? If so, please uninstall and reinstall with the controller disconnected. I've experienced this sort of reaction if the driver is loaded before its ownership is properly set. The OS then refuses to load the driver no matter what until the system is rebooted and a fresh kext is provided. So if the controller is connected, it might be trying to make a connection before the properties are correctly applied.

lpvinyl21, 2018-05-21:
Hi, I did try to reinstall it while the controller was unplugged and still a no go. I think I might reinstall my OS to help some other PC issues as well. Hopefully, that will fix this issue also. 

Thanks for all your help.

FranticRain, 2018-05-21:
Okay. Make sure to let me know if the OS reinstall fixes the issue. It would be great to know for reference for the others with this issue.

Mynasru, 2018-05-28:
I got the exact same issue, I installed it without the controller connected (XBOX360 wireless), running osx 10.13.4