360Controller/360Controller #678: My Xbox One S controller was working but it stopped working

Erallie, 2018-05-18:
My Xbox One S controller was working with my MacOS High Sierra, but it stopped being recognized by the preference pane. I'm using the same cable, and doing the same things, but it just stopped being recognized. I don't know what happened! Please Help!!!
[bitmoji face](https://render.bitstrips.com/v2/cpanel/15521b73-fc60-4035-ad1a-964ebf0ddb24-f5f61c11-d3b6-4b90-b75f-2d15440909ce-v1.png?transparent=1&palette=1&width=246)

Erallie, 2018-05-18:
Thank you for your prompt response. Yes, I checked the first thing and the controller is recognized by the computer. I tried the command in terminal, and I got this message:

/Library/Extensions/360Controller.kext failed to load - (libkern/kext) authentication failure (file ownership/permissions); check the system/kernel logs for errors or try kextutil(8).

I also checked the security and privacy pane and it didn't show anything for me to grant permissions to.

FranticRain, 2018-05-18:
Okay, so this is a "authentication failure" issue. There is currently no fix for this because I haven't figured out what causes it. You aren't the only one with this issue, but honestly, I don't know what is causing it.

Erallie, 2018-05-18:
okay, then. Can you tell me when you find the issue, or should I join another issue to be notified on it?

FranticRain, 2018-05-18:
Don't worry. If I find a solution, you will be notified through this issue.

dsc, 2018-05-21:
while I don't have a lot to contribute here, I'm curious whether these cases somehow involve an ACL (attached to, I would guess `/Library` or `/Library/Extensions`).

what does `sudo /bin/ls -lh@aed /Library/Extensions/360Controller.kext` say, @Erallie? again, this is probably a dead end, but there are only so many sources of "authentication".

dsc, 2018-05-21:
fwiw i get (with no such errors, but using a XB1 S (bluetooth) controller via USB to USB-C cable):

```
drwxr-xr-x@ 3 root  wheel    96B Aug 28  2016 /Library/Extensions/360Controller.kext
	com.apple.metadata:_kTimeMachineNewestSnapshot	  50B
	com.apple.metadata:_kTimeMachineOldestSnapshot	  50B
```

FranticRain, 2018-05-21:
I've tried a `sudo chown -R root:wheel 360Controller.kext` on this issue previously and it didn't resolve the error. I didn't ask OP to try it, but I probably should have checked to make sure. I think Apple's SIP might have a "feature" where a kext loaded with incorrect permissions once will never load again. That might be my imagination, but it seems to be how it reacts to my testing.