360Controller/360Controller #679: Microsoft OEM "Xbox Controller S" Not Functional

mborgerson, 2018-05-24:
Tested on May 24, 2018
Driver Version: v0.16.5
macOS Version: 10.12.6

Connected an original Xbox Microsoft [Xbox Controller S](https://en.wikipedia.org/wiki/Xbox_controller#/media/File:Xbox-s-controller.jpg) via an [Xbox Female to USB-A Male adapter cable](https://www.amazon.com/Mizar-Replacement-Xbox-Controller-Adapter-Microsoft/dp/B000RT2868). Driver appears to be active in enumeration process but fails to correctly handle device.

dmesg output:
```
IOUSBHostDevice: family specific matching fails
IOUSBHostDevice: family specific matching fails
IOUSBHostDevice: family specific matching fails
IOUSBHostDevice: family specific matching fails
IOUSBHostDevice: family specific matching fails
IOUSBHostDevice: family specific matching fails
Xbox360Peripheral::probe(IOUSBHostDevice)
Xbox360Peripheral::start(IOUSBHostDevice) <1>
Unknown device release 120
start - unable to find the interface
Xbox360Peripheral::start(IOUSBHostDevice) <1> failed
```

Thanks,
Matt

FranticRain, 2018-05-24:
I haven't seen @kasbert around in a while, so I guess I'll try and give this a shot. The problem here is that the controller is advertising a different interface than the one we are checking for. In order to fix this, we need to determine what interfaces the controller is advertising. As far as I know, there isn't a way to find this information on macOS. There might be a way to dig it up in Windows, but I know for certain that if you run `lsusb -v` in Linux, then it will give you the information you need about the interfaces of the controller. Once we have that information, then we can fix this issue.

mborgerson, 2018-05-31:
Here's the output from `lsusb -v` via Ubuntu 18.04 with the controller in question (both bus and controller are included):

```
Bus 003 Device 008: ID 045e:0289 Microsoft Corp. Xbox Controller S
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x045e Microsoft Corp.
  idProduct          0x0289 Xbox Controller S
  bcdDevice            1.20
  iManufacturer           0 
  iProduct                0 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        88 Xbox
      bInterfaceSubClass     66 Controller
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               4

Bus 003 Device 007: ID 045e:0288 Microsoft Corp. Xbox Controller S Hub
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            9 Hub
  bDeviceSubClass         0 Unused
  bDeviceProtocol         0 Full speed (or root) hub
  bMaxPacketSize0         8
  idVendor           0x045e Microsoft Corp.
  idProduct          0x0288 Xbox Controller S Hub
  bcdDevice            1.20
  iManufacturer           0 
  iProduct                0 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           25
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval             255
```

FranticRain, 2018-06-01:
So by my best guess, the controller is matching to the hub, not the controller. It is then trying to open the interface on the hub and not the controller, leading to the failure message that you are seeing. I've built a version of the driver that removes the hub, so that the driver should match the the controller instead. I'm not sure that it will fix the issue, but let me know what happens. Sorry that it's in a zip, but github doesn't let me attach pkgs or dmgs.
[Install360Controller.pkg.zip](https://github.com/360Controller/360Controller/files/2061037/Install360Controller.pkg.zip)

FranticRain, 2018-08-12:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-08-12 by FranticRain