360Controller/360Controller #680: Xbox One Controller isn't recognized on Games

James8meout, 2018-05-26:
It shows that the controller works on the driver but in game it doesn't work.

James8meout, 2018-05-26:
It worked before but now it doesnt

FranticRain, 2018-05-26:
The driver does not provide any sort of compatibility for individual games beyond what is provided in the preference pane. The driver simply sends button data to the operating system. Once it does that, control is out of the driver's hands. Since Apple does not provide any standardized means of interacting with controllers in macOS, every game uses a different method to read controller data. We cannot provide individual fixes for each game within the controller. That simply isn't feasible. It is the responsibility of game developers to support controllers, not the other way around. We cannot and will not provide support for individual games. If you are using an Xbox One controller, we can recommend that you use the "Pretend to be an Xbox 360 Controller" option in the "Advanced" menu for better game compatibility. Beyond that, there is nothing we can do.

closed 2018-05-26 by FranticRain