360Controller/360Controller #681: xbox one controller is not being recognized by the driver on Mac

jplumley03, 2018-05-26:
I have a brand new controller and I downloaded the latest driver and my mac is running on the newest OS. When i use a usb and wire it in, "Item 1" will show up in the "device found" box for a split second but then it will say no "no device found" I can't figure out what's going on.

FranticRain, 2018-05-26:
Before we can help you, please consult the README and provide all necessary information so that we can help you with your issue. https://github.com/360Controller/360Controller#xbox-one-controllers-connected-with-usb

jplumley03, 2018-05-27:
my controller does show up in apple hardware usb panel. It isn't showing up in the Xbox 360 controller menu.

jplumley03, 2018-05-27:
it is a new xbox one controller with bluetooth capabilities.

FranticRain, 2018-05-28:
I would suggest trying a different cable. It sounds like the driver is matching to your controller, but something is going wrong and it is losing connection to the controller. I use this driver and that controller perfectly fine together, so I believe that the issue has to be hardware related.

SteveDAC, 2018-05-28:
I am having exactly the same issue. I have an Xbox 360 controller, (can't swap out the cable), and it has been working great up until I attempted to install the driver on a fresh new 10.13.4 macOS installation.

It also shows up in the USB section of System Information.

Also, this may be important, (probably is since I don't believe that the driver can work without it): But when installing at the part where you have to click "Allow" in Security and Privacy preferences, click on that button does nothing.

FranticRain, 2018-05-28:
@SteveDAC So when you say it "does nothing" does that mean that the prompt remains? Does it go away? When you try to reconnect the controller, does the prompt come back? Or does "does nothing" mean that it has no impact on this issue, even after accepted?

If you are having trouble with getting the "accept" button to stay accepted, then consider this bit of information that was recently added to the README: "This prompt has been known to have issues with software or hardware that interferes with mouse movement. If you are using software that impacts the movement of your mouse, or are using a special interface device, such as a Wacom tablet, please using a standard input device, such as a mouse, to click this button. This is a security feature of macOS and is out of our control." Specifically, software or hardware that interferes with normal mouse function needs to be disabled in order to properly accept this prompt.

SteveDAC, 2018-05-29:
Thanks @FranticRain. 

Yes it turns out that that the culprit was an app called MagicKeys used for re-routing the media keys to Spotify.

I really think that changing the behaviour of that button without giving some kind of an indication of why it's not working as expect was a bit of an asshole move by Apply.

FranticRain, 2018-06-23:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-06-23 by FranticRain