360Controller/360Controller #682: No devices found

westeis, 2018-05-26:
I have a brand new Xbox 360 Wireless Controller. 

I made sure to allow the driver on Security & Privacy and still I got nothing. 

Thanks in advance for your reply! 

[image]

FranticRain, 2018-05-26:
I don't think you meant to close this, but the latest release very clearly says that we do not support wireless Xbox 360 controller on versions of macOS 10.11 or later. This is because Apple made changes to the API that they refuse to fix. There is currently no fix for this, but you can get this to work. If you install an old version of the driver before this feature was removed, you can use a wireless controller with your mac. However, your computer WILL kernel panic whenever the computer goes to sleep. If you go into the preference pane and disable the driver before sleeping the computer, then you will not kernel panic.

I am going to close this issue now because it is a duplicate of other issues.

closed 2018-05-26 by FranticRain

westeis, 2018-05-26:
Hi, I'm sorry for that. Yes, I did mean to close the issue because oddly enough, it started to work anyway with this version of the driver. I honestly don't know how.