360Controller/360Controller #684: Xbox 360 Controller works but is blinks

davestarr77, 2018-05-28:
### Type of Controller
Xbox 360

### Connection Method
Wireless Adapter

### Device Name and Info
N/A

### Response of Controller
It connects but blinks slowly

### Response in Preference Pane
Everything works as expected

### Detailed Explaination of Issue
The controller works. It connects and it works in Games. The problem is that since five weeks (haven't changed anything but the normal system updates) it doesn't show the number of the controller with a constant green light (1, light on the top left) but starts blinking slowly when it connects (all four lights). I've tried to deinstall and install again but it doesn't help. I've also allowed the privacy settings for the Developer. Nothing seems to help here.

FranticRain, 2018-05-28:
The driver is supposed to send a special packet to the controller that tells it to turn on a specific light pattern. It appears that it isn't doing that correctly. Unfortunately since wireless controllers are unofficially supported right now, there won't be a fix coming any time soon. I'm going to mark this one as a bug and come back to it at a later date.

davestarr77, 2018-05-28:
Thank you! The thing is that it always worked before. There was just one exception. Maybe this cld. be an indicator: it showed the same behaviour when the computer went to sleep mode and was woken up again. Just a restart cld. fix that.

Sent from mobile device. Please excuse briefness.

DarkCloud14, 2018-05-29:
I had the same problem, I just installed v0.16.4 again (uninstalled v0.16.6 before ofc) and it works again for me. Well I didn't test the sleep mode case but at least for normal use case it works again and it's better than the blinking..

FranticRain, 2018-05-29:
Thanks for the tip. Any idea if it also works in 0.16.5? The only things that should be different between the two are new devices. My guess is that it's a bug introduced in 0.16.6, so if I dig through the diff between 0.16.16 and 0.16.4 should tell me where it is. I don't own any wireless 360 controllers or a receiver, so I can't exactly test it, but I'll give it a shot.

davestarr77, 2018-05-30:
> @FranticRain <https://github.com/FranticRain> I didn't test 0.16.5 until now as a hang was mentioned for macOS 10.13.
> Now I can confirm that it also still works with 0.16.5.

Thank you! Confirmed. 0.16.7 deinstalled, 0.16.5 installed. Works fine.

FranticRain, 2018-06-01:
If you guys get a chance, could you give 0.16.8 a shot? The wireless controller code hasn't been changed in years, so I can't figure out what it is. I assumed that it was some sort of build issue, and the build style has returned to the style of 0.16.5 with the 0.16.8 release. I can't imagine why it would fix it, but it might.

DarkCloud14, 2018-06-01:
@FranticRain well what should I say... it fixed it at least for me with 0.16.8 it works.

FranticRain, 2018-06-02:
Apple's kernel works in mysterious ways, I guess.

closed 2018-06-02 by FranticRain

davestarr77, 2018-06-02:
I don’t know where this email is going but at least I also wanted to confirm that 0.16.8 fixed the issue.