360Controller/360Controller #685: Third Party Controller Found By Mac, Does Not Appear in Preference Pane

MBartner, 2018-05-29:
### Type of Controller
Xbox 360

### Connection Method
Wired

### Device Name and Info
>Device Name: TSZ360 Pad
<br/>Product ID: 0x0201
<br/>Vendor ID: 0x0e6f

### Response of Controller
Guide button lights up once on connection.

### Response in Preference Pane
Does not appear in preference pane.

### Detailed Explaination of Issue
The third party controller is labeled "TSZ360 Pad" and manufactured by Electrosource llc. The device is found in the System Report, but does not appear in the device list in the preference pane. Thank you.

FranticRain, 2018-05-29:
[You should find some instructions on how you can get back to that permission prompt in the README here.](https://github.com/360Controller/360Controller#installation) Let me know if that fixes your issue.

MBartner, 2018-05-29:
Did the trick. Sorry for my incompetence haha. Thank you.

FranticRain, 2018-05-29:
No problem! Glad to hear that it is working.

closed 2018-05-29 by FranticRain