360Controller/360Controller #686: Rock Candy Xbox 360 is not detected by Steam (MAC)

dude396, 2018-05-31:
### Type of Controller
Xbox 360 Rock Candy controller

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Rock Candy Gamepad for Xbox360
<br/>Product ID: 0x011f
<br/>Vendor ID: 0x0e6f

### Response of Controller
The guide button does not light up, and it does not vibrate. 

### Response in Preference Pane
All triggers and joysticks work.

### Detailed Explaination of Issue
Along with the above, the main issue is the controller isn't detected by Steam. The regular Xbox 360 controller (the official Microsoft controller) works with the Mac, but the Rock Candy won't be detected by steam.

FranticRain, 2018-05-31:
The driver does not provide any sort of compatibility for individual games beyond what is provided in the preference pane. The driver simply sends button data to the operating system. Once it does that, control is out of the driver's hands. Since Apple does not provide any standardized means of interacting with controllers in macOS, every game uses a different method to read controller data. We cannot provide individual fixes for each game within the controller. That simply isn't feasible. It is the responsibility of game developers to support controllers, not the other way around. We cannot and will not provide support for individual games. If you are using an Xbox One controller, we can recommend that you use the "Pretend to be an Xbox 360 Controller" option in the "Advanced" menu for better game compatibility. Beyond that, there is nothing we can do.

That being said, the "Pretend to be an Xbox 360 Controller" option should fix you up. If it doesn't. There's nothing we can do. That's Steam's fault. Not ours.

closed 2018-05-31 by FranticRain

dude396, 2018-05-31:
> That's Steam's fault. Not ours.
Is all I needed to hear.
 Apologies if I wasted your time, I'm just looking for help and was wondering if anyone knew a fix.

FranticRain, 2018-05-31:
No problem. We get it all the time.

dude396, 2018-05-31:
The "Pretend to be an Xbox Controller" worked, thank you so much.
Also, I feel a tad like an idiot..

FranticRain, 2018-05-31:
You shouldn't. It's completely nonsensical that it fixes it. Why is Steam looking for the specific vendor and product ID of the Microsoft Xbox 360 controller and not for something more generic?