360Controller/360Controller #689: 360 controller not connecting

KevinFrog, 2018-06-02:
**Type of Controller**
Xbox 360 controller

**Connection Method**
wired connection method

**Device Name and Info**
Vendor-Specific Device:
  Product ID:	0x028f
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	1.05
  Speed:	Up to 12 Mb/sec
  Location ID:	0x14100000 / 2
  Current Available (mA):	500
  Current Required (mA):	100
  Extra Operating Current (mA):	0

**Response of Controller**
No response from controller

**Response in Preference Pane**
no response in preference pane

**Detailed Explanation of Issue**
When I plug the controller in it shows 'No devices found' and there is no other response from the driver

FranticRain, 2018-06-02:
Could you please run `sudo kextutil /Library/Extensions/360Controller.kext` in the terminal for me?

KevinFrog, 2018-06-03:
Baileys-MacBook-Pro:~ baileybear$ sudo kextutil /Library/Extensions/360Controller.kext
Password:
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

FranticRain, 2018-06-03:
Okay, so what this tells me is that the driver is correctly loading. Now, if you could reboot your computer to ensure that the driver has been unloaded and plug in your controller. Then check to see if the driver has been loaded with `sudo kextstat | grep com.mice`. If this returns a value, then the driver has been correctly loaded. If the controller still isn't found in the preference pane, let me know. If you plug the controller in, and that command does not return anything, then run `sudo kextutil /Library/Extensions/360Controller.kext` again and reload the preference pane. Let me know what results you get.

KevinFrog, 2018-06-04:
Baileys-MacBook-Pro:~ baileybear$ sudo kextstat | grep com.mice
Password:
  151    0 0xffffff7f83737000 0xc000     0xc000     com.mice.driver.Xbox360Controller (1.0.0d15) B8A01E48-7B8E-3327-8F36-321B1F346D95 <54 45 5 4 3>

The controller still hasn't been found in the preference pane

FranticRain, 2018-06-04:
Okay. So something is happing in the initialization process that is causing it to fail, most likely. So I need you to open up `Console.app` before you connect your controller to the system. Make sure that your device is selected in the left hand panel, and that in the search bar at the top right, you type in `kernel` and press enter. Here is what it'll look like if you've done it correctly.
[image]
What I need you to do is then plug in your controller, let it sit for a couple seconds, and then copy and paste everything that shows up in the console after you plugged in your controller. Just that first few seconds worth should be plenty, in order to finally track down what your problem is.

My best guess is that you've bought some counterfeit Xbox 360 controller and it doesn't abide by the spec properly. I'll do what I can to help you, but often these controllers can't be made to work.

KevinFrog, 2018-06-05:
default	14:48:39.311234 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.311983 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.311989 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.311995 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312000 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312006 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312011 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312018 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312023 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312028 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312034 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312041 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312046 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312052 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312058 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312063 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312069 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312075 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312080 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312087 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312092 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312098 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312104 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312109 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312115 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312121 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312126 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312133 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312137 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312144 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312149 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312155 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312161 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312166 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312172 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312177 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312183 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312189 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312195 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312200 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312206 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312212 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312217 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312223 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312229 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312235 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312240 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312249 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312255 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312261 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312266 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312272 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312278 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312283 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312289 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312295 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312301 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312307 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312316 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312324 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312333 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312339 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312344 +0930	kernel	IOUSBHostDevice: family specific matching fails
default	14:48:39.312376 +0930	kernel	Xbox360Peripheral::probe(IOUSBHostDevice)
[... cut ...]
default	14:48:41.664897 +0930	kernel	ARPT: 154.359807: txpktpend AC_BK 0 AC_BE 0 AC_VI 0 AC_VO 1 BCMC 0 ATIM 0
default	14:48:55.865865 +0930	kernel	com.apple.p2p: IO80211AWDLPeerManager::setAwdlOperatingMode Setting the AWDL operation mode from AUTO to SUSPENDED
default	14:48:55.865877 +0930	kernel	com.apple.p2p: AWDL is suspended
default	14:48:56.120891 +0930	kernel	com.apple.p2p: IO80211AWDLPeerManager::setAwdlOperatingMode Setting the AWDL operation mode from SUSPENDED to AUTO
default	14:48:56.121150 +0930	kernel	com.apple.p2p: AWDL is off

FranticRain, 2018-06-05:
Okay. So now I know where it failed. This is really curious. Are you sure that this controller is compatible with Xbox consoles? Because if it fails on interface matching, like it did, I'm afraid that it isn't actually a device that the driver can work with. Is this controller something other than an official Xbox 360 wired controller?

The main problem here is that the steps we have to take to discover this interface cannot be done on macOS, at least as far as I know. You'd need a separate computer running Linux, I believe. This may be the end of the line for this issue.

KevinFrog, 2018-06-05:
Yeah, I've been using the controller with my xbox since I bought it without issue. It is an official xbox 360 wired controller. No problem though, I'll sort something out. Thanks a lot for the help!

nathanielwasson, 2018-06-06:
I also have this issue, but my results to the steps above differed from KevinFrog's.

Where my debugging started being different:

> Okay, so what this tells me is that the driver is correctly loading. Now, if you could reboot your computer to ensure that the driver has been unloaded and plug in your controller. Then check to see [...]

`sudo kextstat | grep com.mice` did not return a value

`sudo kextutil /Library/Extensions/360Controller.kext` returned same value again as before:

> Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings:
Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done):
ChatPadKeyboardEvents

Reloading the System Preferences pane resulted in the same issue of 'No devices found'.

KevinFrog, 2018-06-06:
Yep, wired controller

FranticRain, 2018-06-06:
@nathanielwasson Please create a separate issue. Juggling two people with different issues in the same thread is a challenge, especially with the high issue abandonment rates this project has. But, it looks like your controller isn't recognized by the driver. Likely because you have a controller that has a vendor and product ID combination that isn't supported by the driver, or you've done something silly like try to get a wireless Xbox 360 controller to act like a wired one with a "Play & Charge Kit."

@KevinFrog As for you, I am so incredibly baffled. What I really need from you is to run `lsusb -v -d 045e:028f` with the controller plugged in. That will give you output that should have been identical [to the lsbusb output section of this page.](http://free60.org/wiki/GamePad) I've searched high and low for equivalents for macOS and Windows, but I only know how to check it on Linux. Sorry. You might be able to install Ubuntu on a flash drive, boot into it, and then test this, without messing up your current OS installation. That's what I do.

FranticRain, 2018-06-09:
@KevinFrog Are you 100 percent certain that this isn't a wireless controller that has a play and charge kit attached to it? I've been doing some digging and the 0x028f Product ID corresponds to the Xbox 360 wireless controller.

A play and charge kit controller will look like this:
[image]

A wired controller will look like this:
[image]

Are you *absolutely certain* that you have a wired controller, because if you have a wired controller, it is a counterfeit device.

FranticRain, 2018-12-04:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-12-04 by FranticRain