360Controller/360Controller #690: gamepad was connected, but key doesn't work.

o8q, 2018-06-04:
### Type of Controller
Xbox 360 controller. may be...
https://www.amazon.co.jp/gp/product/B074NWW8JS/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1

### Connection Method
wired

### Device Name and Info
製品ID:	0x028e
製造元ID:	0x045e  (Microsoft Corporation)
バージョン:	1.10
速度:	最高12 Mb/秒
製造元:	Xiaoji
場所ID:	0x14200000 / 2
利用可能な電流（mA）:	500
必要な電流（mA）:	500
高負荷時消費電流（mA）:	0

### Response of Controller
No response from controller

### Response in Preference Pane
no response in preference pane

### Detailed Explaination of Issue
I connect gamepad, then xbox 360 Controllers shows tell me 'Xbox 360 Wired Controller (Wired)' was connecting.
and When I push any key(X, Y, or Up...), there is no response in Controller Test Pane.

### Status kext(?)
sudo kextstat | grep com.mice
  155    0 0xffffff7f83b78000 0xc000     0xc000     com.mice.driver.Xbox360Controller (1.0.0d15) B8A01E48-7B8E-3327-8F36-321B1F346D95 <53 47 5 4 3>

### Install Log
Jun  4 14:04:24 iMac Installer[1154]: @(#)PROGRAM:Install  PROJECT:Install-1000
Jun  4 14:04:24 iMac Installer[1154]: @(#)PROGRAM:Installer  PROJECT:Installer-920
Jun  4 14:04:24 iMac Installer[1154]: Hardware: iMac13,2 @ 2.90 GHz (x 4), 8192 MB RAM
Jun  4 14:04:24 iMac Installer[1154]: Running OS Build: Mac OS X 10.13.5 (17F77)
Jun  4 14:04:24 iMac Installer[1154]: Env: __CF_USER_TEXT_ENCODING=0x1F6:0x1:0xE
Jun  4 14:04:24 iMac Installer[1154]: Env: TMPDIR=/var/folders/53/y3tbswyj6d5dd46kpr6jqkww0000gp/T/
Jun  4 14:04:24 iMac Installer[1154]: Env: HOME=/Users/o8q
Jun  4 14:04:24 iMac Installer[1154]: Env: SHELL=/bin/bash
Jun  4 14:04:24 iMac Installer[1154]: Env: SSH_AUTH_SOCK=/private/tmp/com.apple.launchd.UF9wlPykWk/Listeners
Jun  4 14:04:24 iMac Installer[1154]: Env: Apple_PubSub_Socket_Render=/private/tmp/com.apple.launchd.GUplMZIVOg/Render
Jun  4 14:04:24 iMac Installer[1154]: Env: PATH=/usr/bin:/bin:/usr/sbin:/sbin
Jun  4 14:04:24 iMac Installer[1154]: Env: LOGNAME=o8q
Jun  4 14:04:24 iMac Installer[1154]: Env: XPC_SERVICE_NAME=com.apple.xpc.launchd.oneshot.0x1000000d.Installer
Jun  4 14:04:24 iMac Installer[1154]: Env: COMMAND_MODE=unix2003
Jun  4 14:04:24 iMac Installer[1154]: Env: USER=o8q
Jun  4 14:04:24 iMac Installer[1154]: Env: XPC_FLAGS=0x0
Jun  4 14:04:26 iMac Installer[1154]: Install360Controller  Installation Log
Jun  4 14:04:26 iMac Installer[1154]: Opened from: /Volumes/360ControllerInstall/Install360Controller_0.16.8.pkg
Jun  4 14:04:26 iMac Installer[1154]: Product archive /Volumes/360ControllerInstall/Install360Controller_0.16.8.pkg trustLevel=202
Jun  4 14:04:39 iMac deleted[397]: diskmanagement: [DMManager(PrivateMethods) clientConforms:error:]: currentThread=29467=0x731b expectedThread=11523=0x2d03
Jun  4 14:04:43 iMac system_installd[517]: PackageKit: Install sandbox purging reclaimed 0 KB
Jun  4 14:04:46 iMac deleted[397]: diskmanagement: [DMManager(PrivateMethods) clientConforms:error:]: currentThread=14867=0x3a13 expectedThread=11523=0x2d03
Jun  4 14:04:46 iMac deleted[397]: diskmanagement: [DMManager(PrivateMethods) clientConforms:error:]: currentThread=14867=0x3a13 expectedThread=11523=0x2d03
Jun  4 14:04:46 iMac deleted[397]: diskmanagement: [DMManager(PrivateMethods) clientConforms:error:]: currentThread=14867=0x3a13 expectedThread=11523=0x2d03
Jun  4 14:04:46 iMac deleted[397]: diskmanagement: [DMManager(PrivateMethods) clientConforms:error:]: currentThread=14867=0x3a13 expectedThread=11523=0x2d03
Jun  4 14:04:46 iMac deleted[397]: diskmanagement: [DMManager(PrivateMethods) clientConforms:error:]: currentThread=14867=0x3a13 expectedThread=11523=0x2d03
Jun  4 14:04:46 iMac deleted[397]: diskmanagement: [DMManager(PrivateMethods) clientConforms:error:]: currentThread=14867=0x3a13 expectedThread=11523=0x2d03
Jun  4 14:05:01 iMac deleted[397]: diskmanagement: [DMManager(PrivateMethods) clientConforms:error:]: currentThread=25895=0x6527 expectedThread=11523=0x2d03
Jun  4 14:05:02 iMac Installer[1154]: InstallerStatusNotifications plugin loaded
Jun  4 14:05:13 iMac Installer[1154]: Administrator authorization granted.
Jun  4 14:05:15 iMac Installer[1154]: ================================================================================
Jun  4 14:05:15 iMac Installer[1154]: User picked Standard Install
Jun  4 14:05:15 iMac Installer[1154]: Choices selected for installation:
Jun  4 14:05:15 iMac Installer[1154]: 	インストール: "Install360Controller"
Jun  4 14:05:15 iMac Installer[1154]: 		Install360Controller_0.16.8.pkg#Install360Controller.pkg : com.mice.pkg.Xbox360controller : 0.16
Jun  4 14:05:15 iMac Installer[1154]: ================================================================================
Jun  4 14:05:15 iMac Installer[1154]: It took 0.04 seconds to summarize the package selections.
[... 81 lines cut ...]
Jun  4 14:06:43 iMac install_monitor[1166]: Re-included: /Applications, /Library, /System, /bin, /private, /sbin, /usr
Jun  4 14:06:43 iMac installd[515]: PackageKit: releasing backupd
Jun  4 14:06:43 iMac installd[515]: PackageKit: allow user idle system sleep
Jun  4 14:06:43 iMac installd[515]: PackageKit: ----- End install -----
Jun  4 14:06:43 iMac installd[515]: PackageKit: 87.2s elapsed install time
Jun  4 14:06:43 iMac installd[515]: PackageKit: Running idle tasks
Jun  4 14:06:43 iMac installd[515]: PackageKit: Done with sandbox removals
Jun  4 14:06:43 iMac Installer[1154]: PackageKit: Registered bundle file:///Library/Application%20Support/MICE/360Daemon.app/ for uid 502
Jun  4 14:06:43 iMac installd[515]: PackageKit: Removing client PKInstallDaemonClient pid=1154, uid=502 (/System/Library/CoreServices/Installer.app/Contents/MacOS/Installer)
Jun  4 14:06:44 iMac Installer[1154]: Running install actions
Jun  4 14:06:44 iMac Installer[1154]: Removing temporary directory "/var/folders/53/y3tbswyj6d5dd46kpr6jqkww0000gp/T//Install.1154vUGe1n"
Jun  4 14:06:44 iMac Installer[1154]: Finalize disk "hdd"
Jun  4 14:06:44 iMac Installer[1154]: Notifying system of updated components
Jun  4 14:06:44 iMac Installer[1154]: 
Jun  4 14:06:44 iMac Installer[1154]: **** Summary Information ****
Jun  4 14:06:44 iMac Installer[1154]:   Operation      Elapsed time
Jun  4 14:06:44 iMac Installer[1154]: -----------------------------
Jun  4 14:06:44 iMac Installer[1154]:        disk      0.02 seconds
Jun  4 14:06:44 iMac Installer[1154]:      script      0.04 seconds
Jun  4 14:06:44 iMac Installer[1154]:        zero      0.07 seconds
Jun  4 14:06:44 iMac Installer[1154]:     install      88.53 seconds
Jun  4 14:06:44 iMac Installer[1154]:     -total-      88.67 seconds
Jun  4 14:06:44 iMac Installer[1154]: 
Jun  4 14:06:44 iMac Installer[1154]: IFDInstallController 13B60 state = 5
Jun  4 14:06:44 iMac Installer[1154]: Displaying 'Install Succeeded' UI.

FranticRain, 2018-06-04:
This controller is not supported by official Xbox hardware. As such, it's support is outside the scope of this driver. We cannot and will not support this driver.

The reason you are seeing what you are seeing with the driver is because this device advertises itself to computers with the Vendor and Product ID's of a legitimate Xbox 360 controller. It violates the USB HID specification by doing this, and enters the realm of "counterfeit controllers." It would work correctly with the driver if it also imitated the initialization and packet layout of the Xbox 360 controller, but because it's designed to work with smartphones and the PS3, they certainly couldn't do this.

So basically, this device isn't compliant with the standard that this driver supports, but advertises itself like it does. Sorry. This controller is simply out of scope of this driver.

closed 2018-06-04 by FranticRain

o8q, 2018-06-04:
Thank you very much for your reply.
I am going to buy other gamepad which was supported by this driver.(with reading readme or something.)
Thank you!