360Controller/360Controller #691: URGENT stuck on login screen, broke my mac

bachmanity1289, 2018-06-04:
My Mac is now stuck on the login screen with a full progress bar. Happened as soon as I did the restart. 

Saw another person had the same issue but got no support. What a disgrace.

AlexDeweert, 2018-06-05:
Yep same here.

buffaloseven, 2018-06-05:
The solution posted [here](https://github.com/360Controller/360Controller/issues/605#issuecomment-375324761) worked for me. The main issue is that while the new versions of the controller driver don’t contain the wireless receiver left, I still had it in there from older versions. Deleted all the 360-related kexts then restarted, reinstalled the latest driver, and no problems now.

They should probably update the package scripts to remove the wireless receiver kext on incompatible operating systems.

AlexDeweert, 2018-06-05:
@buffaloseven that worked great thanks so much.

FranticRain, 2018-06-05:
@buffaloseven The installers for modern versions of the drivers simply do not contain the wireless driver. The issue is in the symlink between /System/Library/Extensions. It is fundamental for force feedback for wired controllers. This issue with the installer causing issues with launch is actually pretty uncommon, despite the feedback you might see from issues. I'm also especially hesitant to remove this symlink because some systems still hang during install even with it gone. The issue has something to do with the configurations of systems that the driver is being installed onto. Some sort of conflict with other software, not explicitly a bug in the driver.

This issue is a duplicate, so I am going to close it.

closed 2018-06-05 by FranticRain