360Controller/360Controller #692: xbox elite controller not working

camden9, 2018-06-05:
### Type of Controller
xbox one elite controller

### Connection Method
wired with the provided xbox elite controller micro usb
### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: xbox one elite
<br/>Product ID: 0x02e3
<br/>Vendor ID: 0x045e 

### Response of Controller
the controller vibrates when plugged up but does not light up or show up in the driver control.

### Response in Preference Pane
nothing appears in the preference pane, it says no devices found.
### Detailed Explaination of Issue
the controller is not found by the driver, it vibrates and shows up in the system report, but the light does not stay on.

FranticRain, 2018-06-05:
Okay, so we're going to need to do some testing to find out what the issue is here. First, I need you to reboot your system. Once rebooted, open up `Console.app` and ensure that your device's name is selected in the left hand panel. Then enter `kernel` into the search bar. Make sure to press enter, so that it accepts the sort criteria. Then, plug in your controller. A bunch of stuff will show up in the console. I need you to copy and paste all of the stuff that shows up after you plug your controller in. This will help me track down where the issue is.

FranticRain, 2018-08-12:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-08-12 by FranticRain