360Controller/360Controller #695: help please

jackboschma, 2018-06-09:
xbox one afterglow wired controller (has headphone jack)

high sierra 10.13.4

360 controller 0.16.8

i have tried cable that come with the controller and a random micro usb

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: xbox one afterglow 
<br/>Product ID: 048-007-D1 ? or 1118GAA /
<br/>Vendor ID: n/a

the afterglow of the controller lights up. no other response 

no response 

i don't know much about computers and code stuff i just want help to get this to wok so i can play a game n my mac. I have spend probably three hours trying this and using every suggestion  i can find. I have tried restarting my mac book and reinstalling the driver with the controller unplugged. However every time I open the driver it would say no device found. I would really appreciate any help you can give me.

FranticRain, 2018-06-09:
PDP and PowerA Xbox One controllers are incompatible with macOS 10.11+ thanks to a rewrite of the macOS USB kernel. We cannot resolve this issue. It is a bug in Apple's core OS code.

closed 2018-06-09 by FranticRain

jackboschma, 2018-06-09:
Ok thank you.