360Controller/360Controller #699: Not detected xbox one with headphone jack controller

EdgarGlez, 2018-06-11:
### Type of Controller
Xbox one with headphone jack controller

### OS Version
MacOS High Sierra
10.13.3

### Driver Version
0.16.8

### Connection Method
Wireless converter

### Device Name and Info

>Device Name: Controller
<br/>Product ID: 0x02dd
<br/>Vendor ID: 0x045e  

My xbox controller is not being detected by the driver, it vibrates when you connect it but the xbox button won't light up

my Mac detects the controller on the usb with the name "controller"

thank you, please help.

FranticRain, 2018-06-16:
I need you to open up your terminal and run `sudo kextload /Library/Extensions/360Controller.kext` and tell me what it says.

If it says `/Library/Extensions/360Controller.kext failed to load - (libkern/kext) system policy prevents loading; check the system/kernel logs for errors or try kextutil(8).` Then you'll need to allow the driver in the `Security & Privacy` section of your `System Preferences`.
[image]

Otherwise, I'll have to give the error a look to see what the problem is.

EdgarGlez, 2018-06-16:
cool it worked.
thank you very much!

closed 2018-06-16 by FranticRain